wpilibsuite/allwpilib #3174: Java Rotation2d does not implicitly wrap the angle (like c++ does)

CptJJ, 2021-02-14:
The ctor for Rotation2d decalres a value, cos, and sin for the object. When calling .getRadians or .getDegrees, only the value is returned, meaning large angles outside the -180 180 scope will be returned:

  public Rotation2d(@JsonProperty(required = true, value = "radians") double value) {
    m_value = value;
    m_cos = Math.cos(value);
    m_sin = Math.sin(value);
  }

  public double getRadians() {
    return m_value;
  }

The c++ handles this differently by calling the atan2 function on the sin and cos before declaring the value in the ctor:

Rotation2d& Rotation2d::operator+=(const Rotation2d& other) {
  double cos = Cos() * other.Cos() - Sin() * other.Sin();
  double sin = Cos() * other.Sin() + Sin() * other.Cos();
  m_cos = cos;
  m_sin = sin;
  m_value = units::radian_t(std::atan2(m_sin, m_cos));
  return *this;
}

The java version should be modified to have a similar behavior, or have a 2-arg ctor made similar to how 254 handles this in their library:

    public Rotation2d(double radians, boolean normalize) {
        if (normalize) {
            radians = WrapRadians(radians);
        }
        radians_ = radians;
    }

    protected double WrapRadians(double radians) {
        final double k2Pi = 2.0 * Math.PI;
        radians = radians % k2Pi;
        radians = (radians + k2Pi) % k2Pi;
        if (radians > Math.PI)
            radians -= k2Pi;
        return radians;
    }

auscompgeek, 2021-02-14:
> `Rotation2d::operator+=`

That's not the constructor.

CptJJ, 2021-02-14:
> > `Rotation2d::operator+=`
> 
> That's not the constructor.

Not sure about the c++, @calcmogul sent that snipped to me saying the c++ version works as expected.

calcmogul, 2021-02-14:
I said that with regard to RotateBy(), not the constructor.

CptJJ, 2021-02-14:
got it, so I guess this is a feature request for a way to normalize the rotation, considering that calling .rotateby(new Rotation2d()) is the only option

auscompgeek, 2021-02-14:
You can already call MathUtil.angleModulus before constructing your Rotation2d.

prateekma, 2021-03-12:
I think the conversation has been cleared up (none of the constructors normalize the angle but addition (`RotateBy`) does). It's a difficult change to make because it would be silently breaking and there are teams who rely on the fact that it does not normalize when using the constructor.

closed 2021-03-12 by prateekma

godmar, 2026-08-17:
> I think the conversation has been cleared up (none of the constructors normalize the angle but addition (`RotateBy`) does). It's a difficult change to make because it would be silently breaking and there are teams who rely on the fact that it does not normalize when using the constructor.

WPIlib 2027 did [remove](https://docs.wpilib.org/en/latest/docs/yearly-overview/yearly-changelog.html#math) the `m_value` field from `Rotation2d`, so this potential breakage is now occurring.

calcmogul, 2026-08-17:
See https://github.com/wpilibsuite/allwpilib/pull/7490 for more context on why that behavior was removed.