wpilibsuite/allwpilib #3188: Add frc::AnalogEncoder::SetPositionOffset(double)

dawonn, 2021-02-20:
Would be nice to restore an absolute encoder position offset at startup.

PeterJohnson, 2021-02-20:
Is there a particular reason this needs to be done in the library?  I would think it could be easily done in user robot code, eg with a 1-line function in a subsystem.

ThadHouse, 2021-02-20:
This doesn't make a lot of sense, just like we don't have a set position on normal encoders. Either way, when starting, you have to make sure your mechanism is at the same point, and it's probably better to make that the 0 point. 

Do you have a use case where this would actually be preferred over just resetting to zero?

ThadHouse, 2021-02-20:
Ok, after some discussion, we found that there is actually a case for this, along with exposing a ResetRollovers function. Same thing would need to be added to DutyCycleEncoder too. We'll look into it.

dawonn, 2021-02-20:
Thanks! 

The reason I think it makes sense to include this is so that the EnableContinuousInput() bounds are obviously correctly respected rather than needing to wrap the output after applying the offset in the subsystem as well.

We are indeed currently providing this functionality via a subsystem wrapper, but seems pretty simple to add to the API. :)

If there is a brownout or similar in the middle of a match, it could be a catastrophic failure if we need to take time to re-calibrate each device (eg move each device manually to a zero point and reset each offset).

ThadHouse, 2021-02-20:
In a brownout case if the sensor dies, rollovers would be missed. So you can only trust the location within a single rotation. Adding an API like this won't help at all in that case.

dawonn, 2021-02-20:
True.

( Our particular use case is intentionally constrained within a single rotation of the absolute encoder for this reason. )

PR #4052 (2022-02-25): Some API and behavior fixes to absolute encoders
SetPositionOffset was added. Been requested multiple times, and easy to implement.

The javadocs mentioned GetPositionInRotation. It has tripped up many people how to get the absolute position from the encoder (You currently have to have precreated the DutyCycle object). Add this method (as GetAbsolutePostition) to make this easier to do.

The checks for making sure a matching set of values was read was doing direct double comparisions. This worked ok in the DutyCycle case, but has problems in the analog case. Solve this by using an epsilon comp [... cut]
commit bc9e96e8 (2022-02-25): [wpilib] Absolute Encoder API and behavior fixes (#4052)

SetPositionOffset was added. Been requested multiple times, and easy to implement.

The javadocs mentioned GetPositionInRotation. It has tripped up many people how to get the absolute position from the encoder (You currently have to have precreated the DutyCycle object). Add this method (as GetAbsolutePostition) to make this easier to do.

The checks for making sure a matching set of values was read was doing direct double comparisions. This worked ok in the DutyCycle case, but has problems in the analog case. Solve this by using an eps [... cut]
closed 2022-02-25 by PeterJohnson