wpilibsuite/allwpilib #3189: Warn/Throw when trying to write to a read-only SimValue

Starlight220, 2021-02-20:
I don't think this happens and it should if the current behavior is silent no-op.

PeterJohnson, 2021-03-15:
This would be a very breaking change, as SimValue's are also used by the device implementation, and "read only" values are written by the device implementation for reading by simulation code.  We've already renamed "read only" to "direction" to make this clearer.  It is technically possible to "fix" by having the SimValue returned by CreateSimValue be different internally than the SimValue returned by GetSimValue, but I'm not a fan of that unless we actually make them return different types (e.g. they're currently interchangeable and this change would make them non-interchangeable).

Starlight220, 2021-03-15:
I haven't tested this, but what is the current behavior if a read-only SimValue attempts to write values?

Also, where is the implementation of all this? I got lost in the C++ division of classes/files.

PeterJohnson, 2021-03-16:
There's no such thing as a read-only SimValue.  Direction is a state of mind :).  SimValues are always writable.

Source trace:
- Java: https://github.com/wpilibsuite/allwpilib/blob/main/hal/src/main/java/edu/wpi/first/hal/SimValue.java#L41-L43
- JNI: https://github.com/wpilibsuite/allwpilib/blob/main/hal/src/main/native/cpp/jni/SimDeviceJNI.cpp#L221-L226
- C API functions: https://github.com/wpilibsuite/allwpilib/blob/main/hal/src/main/native/sim/SimDevice.cpp#L60-L62
- Next level implementation: https://github.com/wpilibsuite/allwpilib/blob/main/hal/src/main/native/sim/mockdata/SimDeviceData.cpp#L208-L221

Starlight220, 2021-03-17:
So I see that the read-only SimValues take effect only for the SimGUI? In code, they can be written from everywhere. Is this the desired behavior?

PeterJohnson, 2021-03-21:
Direction also shows up in the HAL websockets interface (it determines whether values are prefixed with "<", ">", or "<>").  And yes, I think it's the desired behavior for them to be written in code from anywhere.

Starlight220, 2021-03-21:
If so, it should be documented. The SimDevice field keys don't need the direction prefix, right?