wpilibsuite/allwpilib #3190: Is it normal for DifferentialDriveKinematics to not be perfect?

LudumDareDevelopment, 2021-02-21:
So I've been having some problem with one of our robot's particular robot's drivetrain. The wheels on each side are not in a perfect line, i.e. the center wheels are "pushed" inwards due to the drop-center nature of the drivetrain. This makes it difficult to measure the track width, so instead I used the characterization tool's track width. I don't have access to the graphs right now but I can safely say that the robot is correctly tracking the target left and right velocities very well. I used the onboard SparkMax PID and it seems to be working very well. However, I see that the robot does not follow the trajectory perfectly, especially in the process to get to the end point. The robot follows the target velocities very well, but it does not match the actual desired trajectory correctly, leading me to believe that our robot cannot be modeled perfectly by typical kinematics that produce the desired left and right velocities. I have attached a picture here of the trajectory following graph with Ramsete enabled. Of course, the robot manages to get close enough to the desired endpoint in the end, but in between there is significant cross-track error. With the Ramsete controller disabled, the problem is exacerbated even more. The gap between the robot's trajectory and the desired trajectory is even larger. So my question is: is it normal for kinematics to not model the robot's drivetrain perfectly, and is there a way we could reduce this cross track error with some tuning or some other process to make the robot overlay the desired trajectory better? Any help would be greatly appreciated. Thanks in advance.
[image]

calcmogul, 2021-02-21:
What drivetrain feedforward are you using?

LudumDareDevelopment, 2021-02-21:
I'm using the standard linear drivetrain characterization constants ks, kv, ka. My feedforward object is the following: SimpleMotorFeedforward(0.151, 2.14, 0.543) - ks: 0.151, kv: 2.14, ka: 0.543. I made a custom Ramsete Command that uses the calculated feedforwards in combination with the SparkMax onboard PID (SparkMax PID allows you to pass in velocity feedforwards with the CANPIDController.ArbFFUnits.kVoltage parameter).

calcmogul, 2021-02-21:
A more accurate drivetrain model would use Ks, linear Kv, linear Ka, angular Kv, and angular Ka. The feedforward model you're using now assumes your drivetrain's moment of inertia is 1. If that isn't the case, you can see some tracking inaccuracies on curved trajectories.

LudumDareDevelopment, 2021-02-21:
I see. Thank you very much!

closed 2021-02-21 by LudumDareDevelopment