wpilibsuite/allwpilib #3191: Add port number to AnalogGyro Error Message

dawonn, 2021-02-21:
It would be great if the port number appeared in the error messages to help debug just like the HAL_kInvalidHandle case:

 "HAL: The requested input is not an accumulator channel"

https://github.com/wpilibsuite/allwpilib/blob/959611420b4747b359e7ea15c4a35fa275a61c32/wpilibc/src/main/native/cpp/AnalogGyro.cpp#L148

```
Error at InitGyro [AnalogGyro.cpp:149]: HAL: The requested input is not an accumulator channel
        at frc::AnalogGyro::InitGyro()
        at frc::AnalogGyro::AnalogGyro(std::shared_ptr<frc::AnalogInput>)
        at frc::AnalogGyro::AnalogGyro(int)
        at Drivetrain::Drivetrain()
        at Robot::Robot()
        at void frc::impl::RunRobot<Robot>(std::mutex&, Robot**)
        at frc::StartRobot<Robot>()::{lambda()#1}::operator()() const
        at void std::__invoke_impl<void, frc::StartRobot<Robot>()::{lambda()#1}>(std::__invoke_other, frc::StartRobot<Robot>()::{lambda()#1}&&)
        at std::__invoke_result<frc::StartRobot<Robot>()::{lambda()#1}>::type std::__invoke<frc::StartRobot<Robot>()::{lambda()#1}>(std::__invoke_result&&, (frc::StartRobot<Robot>()::{lambda()#1}&&)...)
        at void std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> >::_M_invoke<0ul>(std::_Index_tuple<0ul>)
        at std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> >::operator()()
        at std::thread::_State_impl<std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> > >::_M_run()
        at 
        at 
        at clone
```

ThadHouse, 2021-02-21:
Ah. This case should be picked up by the PARAMETER_OUT_OF_RANGE case, but we must have changed what error the HAL reported at some point to HAL_INVALID_ACCUMULATOR_CHANNEL. Easy fix to  make, just a note for us on what to fix.

ThadHouse, 2021-05-01:
This will be fixed by #3322

closed 2021-05-01 by ThadHouse