wpilibsuite/allwpilib #3197: [Glass] Add ability to draw on Field2D via Robot Code

Daltz333, 2021-02-21:
This would be incredibly useful for overlying your robots intended trajectory, current trajectory on a field map.

sciencewhiz, 2021-02-21:
Also requested here: https://www.chiefdelphi.com/t/draw-trajectories-in-simulation/390949

webbbt456, 2021-03-09:
Please add the following calls to frc::FieldObject2d

 /**
   * Set multiple points from an array of Translation2d objects.
   *
   * @param points array of 2D points
   */
 void SetLine(wpi::ArrayRef<Translation2d> points);
 /**
   * Set multiple points from an array of Translation2d objects.
   *
   * @param points array of 2D points
   */
void SetLine(std::initializer_list<Translation2d> points);
void SetPolygon(wpi::ArrayRef<Translation2d> points);
void SetPolygon(std::initializer_list<Translation2d> points);

The SetLine calls should display the points by drawing a line from point to point.
The SetPolygon calls should display the points by drawing a filled polygon, it should close the polygon if the points array does not contain the same starting and ending points.

Please do not limit the arrays to 85 points like frc::FieldObject2d::SetPoses.

PeterJohnson, 2021-03-11:
I have a slightly different plan in mind here.  I want to put that flexibility on the display side, rather than the robot code side.  So you can still set an array of poses, and choose to display the poses as a line or polygon instead of images/outlines.

The poses limit is a relatively straightforward fix.

PR #3234 (2021-03-27): [wpilib] Field2d enhancements
- Add raw support for pose lists > 255/3 in length
- Improve drag selection, especially with closely overlapping objects
- Drag selection of corner also highlights center of object with smaller circle
- Multiple styles (box, line, closed line, track)
- Configurable line and arrow settings (color, weight)
- Add tooltip for object name, index, x, y, rotation
- Context menu for pose edit/add/remove
- View/edit in feet or inches as well as meters
- Configurable object selectability

Implementation: use vector of Pose2d internally, use units

Fixes #3197.
closed 2021-03-27 by PeterJohnson