wpilibsuite/allwpilib #3204: Bad imgui.ini File Blocks Glass from Launching

jonathandao0, 2021-02-24:
**Describe the bug**
A bad imgui.ini file can block the Glass window from launching. The simulation terminal window will open and run properly, but the Glass window will fail to launch. The Glass window will continue to fail to launch if you restart the simulation and doesn't clearly point out where the issue is.

**To Reproduce**
Not exactly sure. I think I was opening/closing the windows rapidly where one windows was writing to imgui.ini and another sim process started and interrupted the write.

**Workaround**
Delete the bad imgui.ini file and relaunching the simulator will recreate a new imgui.ini file. This will wipe all your previous settings though.

**Proposed Solution**
Display an error window pointing to the offending section so a less drastic fix can be done manually.

sciencewhiz, 2021-02-24:
Do you still have the bad ini file? It may be fixed by https://github.com/wpilibsuite/allwpilib/pull/3167

calcmogul, 2021-11-09:
Closing this since it's unreproducible.

closed 2021-11-09 by calcmogul