wpilibsuite/allwpilib #3206: Adjust the Weight of the Kitbot Simulation to Something Reasonable, More than 25lbs

jasondaming, 2021-02-25:
When creating a[ kitbot DifferentialDrivetrainSim](https://github.com/wpilibsuite/allwpilib/blob/45590eea220923903606e47c7428939c3ce979e2/wpilibj/src/main/java/edu/wpi/first/wpilibj/simulation/DifferentialDrivetrainSim.java#L483) 25lbs is hardcoded as the weight.  While this weight represents the weight of the AM14U4 it does not take into account: motors, electronics or the battery which are all required to actually run the robot.

prateekma, 2021-02-26:
@jdaming Do you have any ideas in mind? I was thinking around 60_lb?

jasondaming, 2021-02-26:
Yeah I suggested 60lbs earlier.  It will likely be a little less for truly barebones but almost everyone will add something else.

PR #3228 (2021-03-08): Make KoP drivetrain simulation weight 60 lbs
Closes #3206
closed 2021-03-08 by PeterJohnson