wpilibsuite/allwpilib #3219: Simulator: Why is my Field2d object not updating based on a swerve robot code sample?

reteps, 2021-02-28:
**Describe the question you have.**
Hello, is there any way to use a swerve bot with the simulator `Field2d` object?

**Describe the reason for your confusion.**

The online tutorial, https://docs.wpilib.org/en/stable/docs/software/examples-tutorials/drivesim-tutorial/index.html, uses a class `DifferentialDrivetrainSim`, that would not make sense in the context of a swerve bot (I think?).

**Is your question related to a problem? Please describe.**

Using the `SwerveDrivePoseEstimator` example c++ project, I have not been able to get my robot to move inside of the `Field2d` screen on the simulator.

[image]
**Additional context (Modifications made to the example project) **

Added the `Field2d` instance and updated call.

```cpp
// private member variable of Drivetrain
frc::Field2d m_field;
// ....
void Drivetrain::UpdateOdometry() {
  m_poseEstimator.Update(m_gyro.GetRotation2d(), m_frontLeft.GetState(),
                         m_frontRight.GetState(), m_backLeft.GetState(),
                         m_backRight.GetState());

    m_field.SetRobotPose(m_poseEstimator.GetEstimatedPosition());
}
```

Added encoders to prevent overlap inside SwerveModule.
```cpp
SwerveModule::SwerveModule(const int driveMotorChannel,
                           const int turningMotorChannel,
                           const int encoderChannel)
    : m_driveMotor(driveMotorChannel), m_turningMotor(turningMotorChannel), 
    m_driveEncoder(encoderChannel, encoderChannel + 1),
     m_turningEncoder(encoderChannel + 2, encoderChannel + 3) {
  // ...
}
// header of SwerveModule.h
  frc::Encoder m_driveEncoder;
  frc::Encoder m_turningEncoder;
```

I would appreciate some guidance or resources, I am not sure where to go from here.

prateekma, 2021-02-28:
Are you sending the `Field2d` instance over `SmartDashboard`? You'll need the following call, probably in the `Drivetrain` constructor:
```
frc::SmartDashboard::PutData("Field", &m_field);
```

reteps, 2021-02-28:
Oops! I also had that in the constructor, I will edit the details.

```cpp
  Drivetrain() {
      m_gyro.Reset();
      frc::SmartDashboard::PutData("Field", &m_field);
    }
```

PeterJohnson, 2021-02-28:
Where are you calling UpdateOdometry() from?

reteps, 2021-02-28:
I have been updating it inside TeleopPeriodic.
```
  // Robot.cpp
  void TeleopPeriodic() override { DriveWithJoystick(true); m_swerve.UpdateOdometry(); }
```

prateekma, 2021-02-28:
If you print the pose going into `m_field` to cout, does it change as you provide motor inputs? What about the pose in the NetworkTables view under `/SmartDashboard/Field`?

prateekma, 2021-02-28:
Oh, that particular example isn't meant to work in simulation at all. Unfortunately, we weren't able to come up with a stable swerve model (the API would be similar to `DifferentialDriveSim`) in time for the season.

For example, it does not use `EncoderSim` et al. to set encoder positions and velocities in the individual swerve module states. These will always stay at zero so there's nothing in the odometry to update.

reteps, 2021-02-28:
@prateekma Is there any alternatives? Since we are digital so far this season, it has been hard to teach the new students about Swerve Drive. I would also be wiling to help add simulation support to swerve drive.

reteps, 2021-02-28:
@prateekma Would I have any better luck using just the base `SwerveDrive` class and adding the normal `Odometry` object instead?

prateekma, 2021-02-28:
No, that would have the same issue. One thing you could try is to model each swerve module as a single jointed arm (without gravity) for the rotation portion and the driving part as a simple flywheel. Then you can use their associated classes to get reasonable speeds and angles which you feed into the pose estimator/odometry.

I can try to setup something like this tonight and post it here.

reteps, 2021-02-28:
That would be incredibly useful, thank you for your help.

prateekma, 2021-03-01:
I started to work on some of the changes here: <https://github.com/wpilibsuite/allwpilib/compare/main...prateekma:swerve-sim-test>.

There are still a few bugs to work out but I don't think I can work on it more tonight; hopefully I can finish it up this week. It should give you an idea of the direction this is going however.

closed 2023-05-27 by rzblue