wpilibsuite/allwpilib #3243: Loading a PathWeaver Trajectory from file can take 5+ Seconds

jasondaming, 2021-03-16:
**Describe the bug**
Loading a trajectory from the .wpilib.json file that PathWeaver outputs can take a "long" time.  This can/should be done when the robot is started to reduce the effect but loading a significant number of trajectories could dramatically increase startup times.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a new TimedRobot project.
2. Paste this code in robot init or any place that gets run:

3. Paste this code as a method:

**Expected behavior**
Code loads reasonably quickly.  This seems excessively high and maybe it can't be improved but if that is the case then we should document this very well.

Daltz333, 2021-03-16:
In terms of documentation, we already do state that it's recommended to be placed in robotInit() in frc-docs.

calcmogul, 2021-03-16:
The root cause is likely the JSON parser having to convert thousands of strings into doubles. A binary format would be much quicker to load, but that's a breaking export format change.

jasondaming, 2021-03-16:
Attached is the trajectory file.  .txt extension will need to be removed.
[Barrel1.wpilib.json.txt](https://github.com/wpilibsuite/allwpilib/files/6151257/Barrel1.wpilib.json.txt)

jasondaming, 2021-03-16:
@Daltz333 I was thinking of a warning [here](https://docs.wpilib.org/en/latest/docs/software/wpilib-tools/pathweaver/integrating-robot-program.html)

I believe you but where do we mention that the trajectory load will take significant time and should be done in roboInit()?

Daltz333, 2021-03-16:
I guess we don't! Easy enough change if you wanna PR it ;)

jasondaming, 2021-03-16:
I will make the issue and PR it later tonight.

calcmogul, 2021-03-16:
I just tried loading the JSON file in C++ simulation and printing some stuff from it to keep it from being optimized out, and the load took 2ms.

calcmogul, 2021-03-16:
I made a Java program as instructed, and the reported elapsed time is 319ms.

jasondaming, 2021-03-17:
@prateekma tried this in c++ and it seemed to work fine.  What do I need to do to narrow down what is causing this bug?

Daltz333, 2021-03-17:
It may be Jackson.

calcmogul, 2021-03-17:
In particular, it may be the Java class loading latency when Jackson is invoked for the first time. A workaround would be calling some C++ code through JNI to parse the JSON.

Daltz333, 2021-03-19:
@randomstring are you sure your grabbing the trajectory in robotInit()

jasondaming, 2021-03-19:
This sounds like an issue with when you are calling the load.  Here is an example of our [code](https://github.com/6391-Ursuline-Bearbotics/2021_UARobotics_Infinite_Recharge/blob/master/src/main/java/frc/robot/commands/GalacticSearch.java) that calls it on initialization.

@randomstring you need to do `resetOdometry(trajectory.getInitialPose())`

prateekma, 2021-03-19:
After some more discussion, we do think that JNI is the way to go to speed this up; for now, we would recommend reading everything on `robotInit()` to prevent the delay when auto is enabled.

randomstring, 2021-03-19:
Got it. We need to move trajectory parsing from autonomousInit() to robotInit(). Thank you @jasondaming that is a really clean way to incoporate the resetOdometry into the command generation.

PR #3257 (2021-03-21): [wpimath] Use JNI for trajectory serialization
Closes #3243
closed 2021-03-21 by PeterJohnson