wpilibsuite/allwpilib #3245: Compressor Closed Loop Control set to false whenever Test Mode is enabled

CraigFox0, 2021-03-18:
**Describe the bug**
When test mode is enabled on the FRC dashboard, the compressor closed loop is set to false. Even if the robot is disabled and re-enabled with a different mode, it will still be set to false.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable robot containing compressor in test mode

**Expected behavior**
The compressor closed loop control should remain true.

**Desktop (please complete the following information):**
 - OS: Windows
 - Java version 11

AustinShalit, 2021-03-20:
Hi @CraigFox0 can you upload your robot program so we can take a closer look?

CraigFox0, 2021-03-22:
@AustinShalit we tried multiple versions of code on two different robots. We tried [this repository](https://github.com/LSRobotics/2021Robot/tree/e53236e26da910e51ab0876d89783632d1b51787), [this repository](https://github.com/LSRobotics/2019Robot/tree/a6753215f62dada89e008ec4c339970042c77186), and a completely new blank project with only a single solenoid operated by a XboxController.

AustinShalit, 2021-03-24:
Hi @CraigFox0, thank you for the code.

In the constructor of the Compressor object, we add it to the Sendable Registry:

https://github.com/wpilibsuite/allwpilib/blob/10c038d9bfabea683c67465301934ed281f3bd1c/wpilibj/src/main/java/edu/wpi/first/wpilibj/Compressor.java#L40

The implementation of the enabled flag is:

https://github.com/wpilibsuite/allwpilib/blob/10c038d9bfabea683c67465301934ed281f3bd1c/wpilibj/src/main/java/edu/wpi/first/wpilibj/Compressor.java#L207-L216

Instead of `this::enabled` should we be calling `this::getClosedLoopControl`?

CraigFox0, 2021-03-27:
That should work

PR #3269 (2021-03-27): [wpilib] Fix Compressor sendable properties
Closes #3245
closed 2021-03-27 by PeterJohnson