wpilibsuite/allwpilib #3251: [Glass] Add scroll bar when pop-up menu goes off the bottom of screen

dawonn, 2021-03-20:
When using a smaller screen and trying to edit the plots, the bottom of the plot edit submenu goes off the bottom of the screen. I need to move the plot widget to the top of the window to get it to fit, would be nice if a scrollbar appeared when the pop-up was drawn below the bottom of the main window. 

[image]

PeterJohnson, 2021-03-20:
This unfortunately is not easy to fix, as it's how it's implemented in the underlying graphics library.  We can file an issue upstream.  As you noted it can be worked around by dragging the popup to the top of the window (a scrollbar does appear if the overall height is greater than the height of the window).

dawonn, 2021-03-20:
I'll note that it _almost_ works as is. The pop up window will add a scrollbar if its much larger than the window: 

[image]