wpilibsuite/allwpilib #3253: [glass] ignore [MainWindow][GLOBAL] xpos and ypos if that's off screen

dawonn, 2021-03-20:
When using glass on a 4K monitor then opening it on another system with a lower resolution the window can be drawn off-screen with no way to drag it back. Would be nice if glass detected that this condition and opened on screen.  

We have just been setting them to zero when moving between systems, but others might find this difficult.

Daltz333, 2021-03-20:
Possible duplicate of #3241

dawonn, 2021-03-20:
@Daltz333, it's only a partial duplicate. I really don't want to manually create & updated dashboards on multiple systems so sharing the glass.ini file is critical. In this case, 'reset UI' would probably wipe out my configuration too. 

A 'reset gui' option would probably be in a menu somewhere, which would not be clickable if the whole window is off-screen.

Daltz333, 2021-03-20:
I see. That's actually a fairly common problem with lots of applications. If you are windows, you can actually Ctrl+Right click on the window in task bar, select move, and use the arrow keys to move it on screen

PR #3256 (2021-03-21): [wpigui] Ensure window will be initially visible
Only set the initial window position if the window X/Y position is within one
of the connected monitor work areas.

Fixes #3253.
closed 2021-03-21 by PeterJohnson