wpilibsuite/allwpilib #3254: [Glass] [nit] Only write settings to ini if changed from default

dawonn, 2021-03-20:
It would be nice if entries were not added to the ini file for 'default' values. 

For example, there are hundreds of [Data Sources] defined with no name specified which clutters the ini file. True, the vast majority users will not have any need to edit by hand but I regularly copy plots or point plots to identical subsystems in other areas rather than recreate them by hand each time.

Eliminating default values would make the file much easier to maintain / modify. Yeah, its a very minor nit for sure...

AustinShalit, 2021-03-20:
My initial thought on this is that if we change the defaults, that would affect existing projects whereas right now it does not. Not sure if that is good or bad but something to be aware of.

auscompgeek, 2021-03-21:
It seems like changes to the defaults should be considered a breaking change then. I think it would make sense to surface defaults changes at the beginning of a new season out of the box.

PeterJohnson, 2021-11-27:
Done in #3719.

closed 2021-11-27 by PeterJohnson