wpilibsuite/allwpilib #3278: Mechanism2d improvements

PeterJohnson, 2021-04-02:
Currently Mechanism2D requires a JSON file to describe the mechanism, only works in simulation because it uses the HAL SimDevice layer to communicate information, and only allows for a single mechanism display window.  Mechanism2D can be made both more flexible and user-friendly by removing the need for the JSON file and moving the data transport to NetworkTables.  The key is to provide a user-facing API for building out the mechanism components.

The fundamental concept currently implemented in Mechanism2D would remain--a tree structure made up of ligaments with adjustable length and angle to their parent.  Other features could also be added, e.g. loading a background image, adding "handles" for drag and drop, etc, but the focus of this issue is on the API and NetworkTables aspects.

Note: as part of this move, the Mechanism2D class would move from the simulation to the smartdashboard package (similar to the move that happened with Field2d).  For consistency it should also be named Mechanism2d instead of Mechanism2D.

Size/coordinates: Each Mechanism2D has a "draw area" (width and height).  This will be scaled (preserving aspect ratio) to fit the window size on the dashboard.  Identical to Field2d, x and y (0,0) is the lower left corner.

Strawman API:

```
class Mechanism2d implements Sendable:
  Mechanism2d(double width, double height);
  void setBackgroundColor(Color color);
  MechanismJoint2d getRoot(String name, double x, double y);

class MechanismJoint2d:
  MechanismLigament2d getLigament(String name);

class MechanismLigament2d:
  void setAngle(Rotation2d angle);  // angle relative to parent
  double getAngle();
  void setLength(double length);
  double getLength();
  void setColor(Color color);
  void setWeight(double weight);  // line thickness
  MechanismJoint2d getEnd();
```

NetworkTables mapping, starting from the base sendable path, is as follows:
```
/Base/.type: "Mechanism2d" (string)
/Base/backgroundColor: "#RRGGBB" (hex color code, as string)
/Base/dims: [width, height] (double[])
/Base/<rootName>/pos: [x, y] (double[])
/Base/<rootName>/<ligamentName>/type: "line"
/Base/<rootName>/<ligamentName>/color: "#RRGGBB" (hex color code, as string)
/Base/<rootName>/<ligamentName>/angle: angle in degrees (double)
/Base/<rootName>/<ligamentName>/length: length (double)
/Base/<rootName>/<ligamentName>/weight: weight (double)
/Base/<rootName>/<ligamentName>/<ligament2Name>/...
```

Example code that builds an elevator with a wrist:
```
Mechanism2d mech = new Mechanism2d(100, 200);
MechanismLigament2d elev = mech.getRoot("elevator", 50, 0).getLigament("elev");
MechanismLigament2d wrist = elev.getEnd().getLigament("wrist");
elev.setAngle(Rotation2d.fromDegrees(90));
elev.setLength(70);
wrist.setAngle(Rotation2d.fromDegrees(-90));
wrist.setLength(20);
```

gerth2, 2021-04-03:
Is there a way to make a ligament invisible? Would setting length/weight to zero accomplish that?

PeterJohnson, 2021-04-03:
Probably the most logical way would be setting the weight to zero.  Especially if you want to be able to have a “gap” in the model.

PR #3281 (2021-04-30): Mechanism2.0d
Fixes #3278
closed 2021-04-30 by PeterJohnson