wpilibsuite/allwpilib #3282: PIDController's m_period

lidor51, 2021-04-04:
Why PIDController's m_period is final? There are cases that I want to use the same pid controller, with different dt gaps.

**Describe the solution you'd like**
Add a calculate(double measurment, double dt) overloading method.

sciencewhiz, 2021-04-04:
Can you describe your use case for having different dt?

lidor51, 2021-04-04:
1. It should be more precise to use exact dt.
2. I'm doing a heading controller (specifically to swerve's heading during teleop driving), and sometimes want to disable the controller. One other solution that I've is to reset the pid controller when I temporary disables it, but doesn't sure if it'll get better results than calculating with a new dt.

sciencewhiz, 2021-05-18:
For 1, have you measured how much dt changes and the impact that has on the controller?

I'm confused by 2. Are you saying that you'd stop calling it for a long time, and then pass the long time? Lets say that you have integral control and while control was disabled the object moved. You'll have huge integral windup because of the large dt. That's why you should reset. Are are you describing a different scenario?

Daltz333, 2021-07-04:
Any update on this @lidor51 ?

calcmogul, 2021-07-11:
> 1. It should be more precise to use exact dt.

Rediscretizing the controller every timestep with a jitter of like +- 2ms is excessive and unnecessary. You won't notice a performance difference.

> 2. I'm doing a heading controller (specifically to swerve's heading during teleop driving), and sometimes want to disable the controller. One other solution that I've is to reset the pid controller [...]

You can "disable" the controller by just not calling calculate(). It won't continue integrating while you're not calling it, which is the desired behavior, in my opinion. The next update won't cause a large jump in the integral either since it just uses `m_period`:
```java
      m_totalError =
          MathUtil.clamp(
              m_totalError + m_positionError * m_period,
              m_minimumIntegral / m_ki,
              m_maximumIntegral / m_ki);
```

If you're relying on integral control too much, I'd suggest using a ProfiledPIDController instead so you can take advantage of a feedforward. That'll drastically reduce steady-state error.

closed 2021-08-30 by calcmogul