wpilibsuite/allwpilib #3289: Wpilib toolchain on Ubuntu Focal?

LinuxRocks2000, 2021-04-06:
**Describe the question you have.**
Attempting to install the WpiLib toolchain on Ubuntu Focal, but it doesn't have a Release file.

**Describe the reason for your confusion.**
I use Linux and don't want to use Windows

**Is your question related to a problem? Please describe.**
It frustrates me when WpiLib doesn't have easily found documentation on GNU/Linux features

**Additional context**
Using Ubuntu 20.10 with Conductor DS and the Wpilib Project Generator utility

calcmogul, 2021-04-06:
The installer puts the toolchain in `~/wpilib/2021/roborio/bin`. You can add that location to PATH.

Daltz333, 2021-04-06:
Additionally Linux is less than 5% of our user base. It's not a main target platform.

LinuxRocks2000, 2021-04-06:
@Daltz333 And yet, Linux peoples exist :)

Daltz333, 2021-04-06:
@LinuxRocks2000

We publish a installer as Tyler linked.

LinuxRocks2000, 2021-04-06:
@Daltz333 I know. Smiley was not a typo.

LinuxRocks2000, 2021-04-06:
Also how do I use the toolchains after they are installed? The documentation does not explain that.

calcmogul, 2021-04-06:
> Also how do I use the toolchains after they are installed? The documentation does not explain that.

I explained this in my earlier reply.

LinuxRocks2000, 2021-04-06:
Oh ok

calcmogul, 2021-04-06:
We have the giant installer to reduce the variation in installation setups for all our supported platforms. If you want to install just the toolchain, you can extract the .tar.gz from https://github.com/wpilibsuite/roborio-toolchain and add it to PATH. You won't get any of the other tools that way though.

Given how small our Linux userbase is, it's not worth maintaining separate packages for every Linux distro. Maintaining apt packages alone was a giant pain because the whole packaging architecture is, to be honest, a disaster.

closed 2021-04-06 by calcmogul