wpilibsuite/allwpilib #3307: HolonomicDriveController & SwerveControllerCommand

lidor51, 2021-04-19:
**Describe the question you have.**
Hi,
I got 2 questions related to the HolonomicDriveController & SwerveControllerCommand.
1. 
* Why does the HolonomicDriveController's correction is field relative? (And not Robot relative)
* In practice, there's any reason should I use different P value for X and Y? (I understand the reason for different PID controllers, but does not understand the reason of using Px != Py  for those controllers, at least in a field relative correction).

2. Can I controller heading independently while following XY trajectory, using the SwerveControllerCommand?

**Describe the reason for your confusion.**
1. In case that HolonomicDriveController's correction is field relative, I don't find any reason to have a different P for X and Y,
    (If i'll have different Px and Py and run the same trajectory but rotated, I'm expecting to find a little bit different behavior, see 
    attached example).
    To me, it's sounds more reasonable to make the correction Robot's relative, which will justify having Px != Py,
    Since Px will represent the forward correction the robot should do, and Py will represent strafe correction. In that case the user 
    can manage the forward & strafes corrections.
2. From looking over the code, it seems that SwerveControllerCommand will in fact always control Trajectory's XY and desired 
    heading ("m_desiredRotation") that can be independent from the trajectory.
    The confusion is that in this discussion they came up to conclusion that it doesn't possible with WPILib.
    [https://www.chiefdelphi.com/t/swerve-trajectory-tracking-with-heading-control/392018/15](https://www.chiefdelphi.com/t/swerve-trajectory-tracking-with-heading-control/392018/15)

**Is your question related to a problem? Please describe.**
No.

**Additional context**
1. Attached example for Q1. Same path, rotated 90 degrees. With Px != Py, I'm expecting  different behavior in each execution of this trajectory, although it's in fact the same trajectory.
[image]

prateekma, 2021-04-19:
> Why is the HolonomicDriveController correction field relative?

There's no particular reason for this, other than it made the implementation easy. In reality, it shouldn't matter whether we are correcting for field-relative error or robot-relative error because everything gets transformed back into individual module speeds in the end. 

> In practice, should I need to use different feedback gains for X and Y controllers?

No, you don't need to. You wouldn't need to if the error was robot relative either because the behavior is essentially the same with either--the robot hardware doesn't know which way is forward.

> Can I control the heading independently using `SwerveControllerCommand`?

Yes. There's a constructor parameter that takes a `Supplier<Rotation2d>` for this purpose.

> The confusion is that in [this discussion](chiefdelphi.com/t/swerve-trajectory-tracking-with-heading-control/392018/15) they came to the conclusion that it isn't possible with WPILib.

That conclusion is incorrect. However, the last post in the thread seems to reinforce what I stated earlier that you can specify headings independent of the defined trajectory and the robot will follow them.

> Same path rotated 90 degrees. With Px != Py, I'm expecting different behavior in each execution of this trajectory, although it's in fact the same trajectory.

I'm not sure I understand exactly what is happening in the graphs you provided. If that is being run in simulation, you will get pretty good tracking regardless of your feedback gains if your feedforward is accurate. Changing the feedback gains for X and Y should not substantially affect tracking performance given an accurate feedforward.

lidor51, 2021-04-19:
Hi, first of all thanks for the response.
> The confusion is that in [this discussion](chiefdelphi.com/t/swerve-trajectory-tracking-with-heading-control/392018/15) they came to the conclusion that it isn't possible with WPILib.

In the discussion they talk "Swerve4ControllerCommand" which I'm not familiar with, but anyway glad that the confusion is in the discussion. BTW, there's a way in WPILib to create a trajectory that also the endpoint is Translation2d (I want the smoothest path the end in that Translation no matter the heading, because I'll correct my heading independently of the trajectory).
(I don't care if the startpoint will be a Pose or a Translation). 

> There's no particular reason for this, other than it made the implementation easy. In reality, it shouldn't matter whether we are correcting for field-relative error or robot-relative error because everything gets transformed back into individual module speeds in the end.
Thanks, but it wasn't my intention, I understand what goes in the HolonomicDriveController.

I'll try to rephrase:
Why it's not implemented like that:

[image]
Marked in yellow, variables that I've change their names, to demonstrate my intention.
Marked in teal, that code that I actually changed.

* forward and sideways mean to be relative to the direction of the path.
Now the correction is based on the error, "forward error" determine how much the robot is early/late,
and "sideways error" determines how much the robot is right/left off the path.

That's useful because let's say I care more about ending (or in general following) correct in right/left direction than forward/backward, so I'll tune each of their kP accordingly.

*if you prefer, here's also the code as text:
  public ChassisSpeeds calculate(
      Pose2d currentPose, Pose2d poseRef, double linearVelocityRefMeters, Rotation2d angleRef) {
    // Calculate feedforward velocities (field-relative).
    double forwardFF = linearVelocityRefMeters;
    double sidewaysFF = 0.0;
    double thetaFF =
    m_thetaController.calculate(currentPose.getRotation().getRadians(), angleRef.getRadians());
    
    m_poseError = poseRef.relativeTo(currentPose);
    
    if (!m_enabled) {
      return ChassisSpeeds.fromFieldRelativeSpeeds(forwardFF, sidewaysFF, thetaFF, m_poseError.getRotation().unaryMinus());
    }

    // Calculate feedback velocities (based on position error).
    double forwardFeedback = m_xController.calculate(0.0, m_poseError.getTranslation().getX());
    double sidewaysFeedback = m_yController.calculate(0.0, m_poseError.getTranslation().getY());

    // Return next output.
    return ChassisSpeeds.fromFieldRelativeSpeeds(
        forwardFF + forwardFeedback, sidewaysFF + sidewaysFeedback, thetaFF,  m_poseError.getRotation().unaryMinus());
  }

Now I'll return to the figure:
[image]
With the original implementation, apart from that in case that pX != pY, we'll observe a different behavior,
we also can't control our correction relative to direction of the path.

An actual example in which using the suggested implementation (correction based on errors that relative to path's direction) be beneficial:
In the end of a trajectory use sideways_error from a Vision, and continue controlling forward_error ( m_poseError.getTranslation().getX()) with the trajectory.

lidor51, 2021-04-19:
For my last comment, please read the second edit in the history instead of the recent (a suggested code is attached).

lidor51, 2021-04-20:
> No there is not. I'm not sure what heuristic we would use to determine "smoothness". It's probably best to play around in Pathweaver and determine what looks best.

I might pre check some cases with pathweaver to get a feeling, but ultimately I want to use it to create on-the-fly trajectories during teleop.

> If you want to incorporate vision tracking, I would recommend that create your own controller and adjust custom strafing behavior there.

Thats probably what we will do. 

> will get very good tracking of the reference trajectory. 

Yeah I'm sure that the current method handles a good trajectory, just want a one all around solution for all our cases (with vision, without, etc).

> It would also be a difficult change to make because it would silently break everyone's PID gains.

Make sense. We also don't want to have double gains, one for current and one for proposed, so we might just override the calculate method, and use only forward/strafe solution.

lidor51, 2021-04-20:
I've looked at those two methods of HolonomicDriveController:

```
  /**
   * Returns true if the pose error is within tolerance of the reference.
   *
   * @return True if the pose error is within tolerance of the reference.
   */
  public boolean atReference() {
    final var eTranslate = m_poseError.getTranslation();
    final var eRotate = m_poseError.getRotation();
    final var tolTranslate = m_poseTolerance.getTranslation();
    final var tolRotate = m_poseTolerance.getRotation();
    return Math.abs(eTranslate.getX()) < tolTranslate.getX()
        && Math.abs(eTranslate.getY()) < tolTranslate.getY()
        && Math.abs(eRotate.getRadians()) < tolRotate.getRadians();
  }

  /**
   * Sets the pose error which is considered tolerance for use with atReference().
   *
   * @param tolerance The pose error which is tolerable.
   */
  public void setTolerance(Pose2d tolerance) {
    m_poseTolerance = tolerance;
  }
```

Seems that unless we're in the specific case in which `angleRef == currentPose.getRotation()`, this line:`Math.abs(eRotate.getRadians()) < tolRotate.getRadians()` of "atReference" method would not represent the real reference we're attempting to reach. 
As in the rotational dimension, we attempt to follow `angleRef`, the rotational error should be the error between `angleRef `and `currentPose.getRotation()`.

Although HolonomicDriveController/SwerveControllerCommand aren't using atReference at all, yet I believe this method should represent the state relative to the real reference we're trying to reach.

prateekma, 2021-04-20:
> Can you make the HolonomicDriveController fields protected instead of private?

I think this is something that we can consider, but typically this is not something we've done in the past before. You can try commenting on [this disussion](https://github.com/wpilibsuite/allwpilib/discussions/3237) to get more feedback.

> Can you have a generic SwerveControllerCommand that takes any drive controller which has [...] method signature?

Once again, I think this introduces another layer of abstraction (we would have to introduce a HolonomicDriveController interface) which I don't really think belongs in WPILib. 

SwerveControllerCommand (like RamseteCommand et. al.) are meant to be full-fledged solutions for very common problems and are not designed to be customizable. I don't think introducing more tuning knobs into these already (confusing for some) classes is a good idea.

> HolonomicDriveController atReference()

I believe you're looking at an older version of the code. The latest code on the `main` branch does this correctly: https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/src/main/java/edu/wpi/first/wpilibj/controller/HolonomicDriveController.java#L58

lidor51, 2021-04-21:
> I think this is something that we can consider, but typically not something we've done in the past before. You can try commenting on [this disussion](https://github.com/wpilibsuite/allwpilib/discussions/3237) to get more feedback.

That's will be great. I believe that WPILib framework doesn't intend to **only** provide a full-solutions, but also be an educational tool that helps students to develop code, and not only consume it.
I will post there as well. Why is that something the didn't happen before?

> SwerveControllerCommand ... I don't think introducing more tuning knobs into these (already confusing for some) classes is a good idea.

That's actually won't change anything to those who still want to use it as a full-fledged solution. The current c'tors will still be exist, it only require to add another c'tor for those who do want to continue develop and extend your tools.

> we would have to introduce a HolonomicDriveController interface

I don't think the cost will be high, and in general I believe that's healthy to put all other Drive Controllers under the same interface.

In general, I believe the price we all pay for not doing changes like this will be higher for some reasons 
(duplicating and changing manually parts of WPILib's code, or just don't using WPILib tools).

closed 2021-07-11 by calcmogul