wpilibsuite/allwpilib #3308: Raspberry pi zero

Phosvan, 2021-04-22:
**Describe the question you have.**
What is the absolute minimal I need to get SPARKMAX libs working 

**Describe the reason for your confusion.**
for weeks I have went from no knowledge of this to reverse engineering CAN, to sending messages via pycan to make the dc motor turn. I want to use all the library's provided in c++. (I am new to this whole thing so I'm ignorant. ) I figured if I went and grabbed all the include header files from FRC and FRC2 and wpi exc.... then include them in the proper location on the raspberry pi zero, the /usr/include folder. I used gcc, told me I need to use 17+ (not sure on the exact number) so I did. and I got 300 errors. ill include a pastbin at the bottom of the thing. It has been a struggle the WHOLE way. I cant cmake the thing, I cant gradel the thing. all I want is to use g++ and minimal c++ files. The exact error is below. I have probably spent so many hours on all three paths over the past month. I do not understand why this is so incredibly difficult or what im doing wrong. Absolutely everything seems to be difficult. I would like to use the c++ lib as it has so much more potential than the PWM I'm currently using. is it posable to retain function by just having files in the include folder? or am I being ignorant? 

Thank you,

**Additional context**
https://pastebin.com/WVvWJES9

ThadHouse, 2021-04-22:
You likely need to enable C++ 17 mode in the compiler with `-std=c++17 `. GCC defaults to C++ 14 mode, which won't have those functions.

Phosvan, 2021-04-22:
WOAH, thanks for the fast reply! I did try this a week or so ago, and I got this https://pastebin.com/3D72ZakX That lead me on some wild rabbit hole that I never figured out

ThadHouse, 2021-04-22:
You need to be using `g++` there, not `gcc`. `g++` includes the C++ libraries in the link list, whereas `gcc` does not.

ThadHouse, 2021-04-22:
And then you're going to start getting linker errors for the WPILib libraries. Those are installed by default on the raspberry pi image, I'm just not sure where. There should be some docs on frcdocs on where those libraries are.

Phosvan, 2021-04-22:
Ah! I have also tried that, but for the sake of being sure I did it again, this is the error I got 
https://pastebin.com/i4GbCytz

which is what I believe the linker errors are?
Does my method seem to have some validity to it? like am I on the right track?
I will work on this tonight I'm studying for finals.
 Thanks for the help! and its super cool something you worked on was used on the mars drone

ThadHouse, 2021-04-22:
Yup, now youre at the point where you need to also add the wpilib libraries to your compile list so they can be linked to. Once you have that, you should be good to go.

Phosvan, 2021-04-22:
For the last 2 hours I have attempted to figure out what you mean. The best I can figure is I need the CPP files too. For an example ill use wpiutil. Under src/main/native/cpp there are alot of cpp files, and one a couple header files. now if I look under include, i get the WPI folder which is what I have included in my linux /usr/include. If i look into libuv there is a another whole src, include file structure. From attempting to run Robot.cpp I have grabbed all the required headers, BUT NO .cpp files. Is this why I am having these issues, and where do I place those .cpp files?
[image]
This is what I'm adding to the raspberry pi in order to run Robot.cpp

auscompgeek, 2021-04-22:
I would not recommend building WPILib yourself, especially by hand, if you don't need to. Use the Maven artifacts like you would be for the Spark Max libraries.

Phosvan, 2021-04-22:
I have no idea how that works, and I do not wanna overstep my welcome here. I looked at the MavenArtifacts.md file, read it 3 times, had absolutely no idea what to make of it. Went to the link https://frcmaven.wpi.edu/artifactory/release/ then clicked on every file for wpiutil I went to release/edu/wpi/first/wpiutil/wpiutil-cpp/2021.2.2 then downloaded wpiutil-cpp-2021.2.2-headers.zip . I repeated this for every directory. I have not the slightest idea how maven would do this. 
[image]
my include file now looks like this. 

For a moment I believed it was because of #include<> vs include"" , however, regardless of where the files are, 
[image]
I still get a reference error.  https://pastebin.com/fVBcr7jZ . 

you state that "if building with static dependencies, the listed order should be the link order in your linker."  I honestly have no idea what to make of that. 

Am I missing a namespace? it is an undefined reference to rev::xxx and frc:xxx  I have moved files around as I thought it was because the files being in other files was the issue. it was not.

I know this kind of may be all over the place, I just cant seem to get absolutely anything to work and you have been super helpful.

Thank you

ThadHouse, 2021-04-22:
In C++, just the headers are not enough to actually build a project. You would need the actual binaries for your platform to link to. So you would need to grab all the linuxraspbian artifacts from maven, and the linuxraspbian artifacts for the sparkmax, which will give you a bunch of .so files that you then need to pass to the compiler.

The issues you're seeing are not a code issue, but instead the compiler just can't find the implementations of whats being used. The actual libraries in C++ are stored separately from the headers (in most cases, there are exceptions) so you have to have the libraries to link to in order to actually finish the build.

Phosvan, 2021-04-24:
I am still out here struggling. I have been doing this for like 2 days and no matter how I approach getting this to work, I just cant get it. I got kind of far with gradle once I found a java 11 for the rpizero processor. 
https://pastebin.com/gCgaGaHi

Honestly, at this point just tell me how you would go about this because I know I am doing this so incredibly stupid. I have only ever used g++ and this is an incredibly steep learning curve for me. Although, I did build something with make (NOT CMAKE. Cant get that to work on rpi zero either) today and it made me quite happy.

Thank you

closed 2021-11-09 by calcmogul