wpilibsuite/allwpilib #3321: wpigui: Simulator GUI launches and crashes on Apple Silicon through Rosetta 2 due to GLFW Cocoa Error (patched in updated GLFW)

richiksc, 2021-04-29:
**Describe the bug**
The WPI Simulator GUI will launch and the window will open but then it will crash quickly.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `./gradlew simulateJava`
2. GUI opens and crashes
3. Run generated `gradlew_simulateJava.sh` shell script
4. View log output

```
HAL Extensions: Attempting to load: libhalsim_gui
Simulator GUI Initializing.
GLFW Error 65544: Cocoa: Failed to find service port for display
Simulator GUI Initialized!
```

**Expected behavior**
Simulator launches and stays open.

**Desktop (please complete the following information):**
 - OS: macOS 11.2 Big Sur, aarch64 (M1)
 - Java version 15.0.1, x64

**Additional context**
After some quick googling, it looks like this issue was patched in https://github.com/glfw/glfw/pull/1833. GLFW version 3.3.3+ should have the fix.

prateekma, 2021-05-11:
Closed in #3344

closed 2021-05-11 by prateekma

richiksc, 2021-05-18:
Any chance this could get released in a 2021 update release? Would be very beneficial to have for this season.

prateekma, 2021-05-21:
It is unlikely that we will do another 2021 release, especially given that we need to create separate branches and tag those (and we haven't tested the Actions infrastructure on branches). It was an unsupported platform for 2021 and I hope you understand that. We are working hard to make sure that Apple Silicon Macs are supported for 2022.