wpilibsuite/allwpilib #3330: Can l use the "Check for WPILIB updates" to update my WPILIB directly

mendax1234, 2021-05-02:
**Describe the question you have.**
A clear and concise description of what you want clarification on.

Can l use the "Check for WPILIB updates" to update my WPILIB directly?

**Describe the reason for your confusion.**
A clear and concise description of why the question requires clarification and what's confusing.

l have this question because l don't want to uninstall my old version of WPILIB and l just want to know whether it can be upgraded from the software itself.

**Is your question related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

None.

**Additional context**
Add any other context or screenshots about the question here.

None

ThadHouse, 2021-05-02:
You don't need to uninstall old versions in order to install the new versions through the installer. You can just rerun the new installer and it will safely install in place.

We recommend doing that over the Check for WPILib updates command in VS Code. That will ensure all dependencies are properly installed, and that the tools get updated, which doesn't happen when you do Check for WPILib updates. One more clarification, Check for WPILib updates only updates in the current year. It won't update from a 2020 project to a 2021 project for instance.

mendax1234, 2021-05-02:
> You don't need to uninstall old versions in order to install the new versions through the installer. You can just rerun the new installer and it will safely install in place.
>
> We recommend doing [...]

Thanks a lot and will it cover my old version automatically ? Or it will take two times of storage than before .

mendax1234, 2021-05-02:
> You don't need to uninstall old versions in order to install the new versions through the installer. You can just rerun the new installer and it will safely install in place.
>
> We recommend doing [...]

l have already used the Update command in the vscode but some of the version is still 2021.2.2 and that's wgy?

ThadHouse, 2021-05-02:
It will double up the artifacts, but it won't double install things like the compiler or vscode.

The update command only updates the project. It doesn't update the extension or any of the tools.

mendax1234, 2021-05-02:
So,the installer is the recommend solution?

ThadHouse, 2021-05-02:
Yes

mendax1234, 2021-05-02:
Thanks a lot！

ThadHouse, 2021-05-02:
If you have any more questions feel free to ask, but I'm going to close the issue.

closed 2021-05-02 by ThadHouse