wpilibsuite/allwpilib #3338: Make GenericHID non-abstract

Daltz333, 2021-05-08:
Joystick class itself is intended for use with Flight Joysticks
XBoxController is intended for use with XBox controllers
GenericHID should be usable for all others.

While Joystick class is useful for the purposes of GenericHID, the naming is technically improper for usage of non-flightstick controllers as they are not "joysticks"

Making GenericHID non-abstract is a free and cheap way of solving this inconsistency.

Starlight220, 2021-05-11:
I'm of the opinion to move the `getY(Hand)` and similar methods to `XBoxCntroller`, and the `getY()` and similar to `Joystick`. The only methods in `GenericHID` should be the `getRaw()` ones.

terbos90803, 2021-05-11:
I completely agree with this.  Put the getY(Hand) and getY() methods only where appropriate.
And while you're in there...
The two most popular game controllers are the XBox and F310.  Only the XBox has a proper class.  The F310 can emulate the XBox to some extent, but has slight differences.
Would it be possible to separate XBox and F310 as two classes?  Or are they truly close enough that they can be converged into a single "GameController" class?

Starlight220, 2021-05-11:
My team (and many others I've seen) use PS4 controllers too, so I'm thinking that we should add that too.
I think that having 2-3 separate classes for it is too much as they differ only in channel indexes (AFAIK), and I think a more appropriate solution would be a `ControllerType` (name TBD) enum that sets the channel indexes according to that preset; then the `GameController()` ctor takes that enum.

auscompgeek, 2021-05-12:
> The F310 can emulate the XBox to some extent, but has slight differences.

Is it really any different software-wise? Wasn't XInput supposed to basically make all gamepads uniform?

Starlight220, 2021-05-12:
There are still some differences as to channel mappings, nonexisting buttons on the triggers and sticks, etc.

PR #3345 (2021-08-14): [wpilib] Split GenericHID functionality
Fixes #3338 

This PR accomplishes the following changes:

- GenericHID is now concrete, and has only `getRawAxis/Button(int)` functionality.
- `getXxx()` has been moved into `Joystick` as that's the only place where it makes sense.
- `Hand` (and therefore `getXxx(Hand)`) has been removed, replaced by specific `getLeft/RightXxx()` methods in `XboxController` and the new `PS4Controller` class.
- C++ `::Button::` and `::Axis::` enums have been converted to identically-namespaced `static constexpr int`s

Obviously, most of these changes are breaking.

---
O [... cut]
closed 2021-08-14 by PeterJohnson