wpilibsuite/allwpilib #3349: MechanumControllerCommand and SwerveControllerCommand examples broken

willtoth, 2021-05-12:
**Describe the bug**

Both MechanumControllerCommand and SwerveControllerCommand examples throw two separate runtime exceptions on initialization.

The first one is due to duplicate resource allocation, both examples try to allocate DIO 5 for encoders:

https://github.com/wpilibsuite/allwpilib/blob/main/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/swervecontrollercommand/Constants.java#L33-L34

The second one is due to the default command not having its own subsystem as a requirement. For example

https://github.com/wpilibsuite/allwpilib/blob/main/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/swervecontrollercommand/RobotContainer.java#L52-L56

Should be

```java
m_robotDrive.drive(
    m_driverController.getY(GenericHID.Hand.kLeft),
    m_driverController.getX(GenericHID.Hand.kRight),
    m_driverController.getX(GenericHID.Hand.kLeft),
    false), m_robotDrive));
```

prateekma, 2021-05-12:
Part 1 is a duplicate of #3089.

sciencewhiz, 2021-05-14:
This is not a duplicate of 3089. 3089 is for SwerveBot, this is for SwerveControllerCommand. Additionally, SwerveBot needs the encoder paramertarized. SwerveControllerCommand already has it parameratrized but uses the same pin in two places.

PR #3359 (2021-05-16): [examples] Fix Swerve and Mecanum examples
Fix encoder allocation and default command.
Fixes #3349
commit 1873fbef (2021-05-16): [examples] Fix Swerve and Mecanum examples (#3359)

Fix encoder allocation and default command.
Fixes #3349
closed 2021-05-16 by PeterJohnson