wpilibsuite/allwpilib #3350: Discussion for deprecation of RamseteCommand.

Daltz333, 2021-05-12:
RamseteCommand has proven more of a footgun to users than beneficial in many proven cases. It only takes one feedforward, assuming that both wheels are the same, but this has proven not true on some unstable drivetrains (Romi is often like this due to bad tolerances).

Pros:
- Ease of Use for users
- Allows easy use with other command convenience functions

Cons:
- Makes it harder to debug projected wheel speeds (which is very critical for fine tuning PIDF

Issue opened on behest of @PeterJohnson

Starlight220, 2021-05-12:
I think that while RamseteCommand has its issues, the ease of use is the only thing granting lower-level teams the ability to use trajectory tracking. It violates the single-concern design principle, so IMO it should be dissected while keeping an easy-to-use option.

calcmogul, 2021-05-12:
The problem is it doesn't really help lower-level teams all that much. It either works out of the box, or it doesn't and is completely undebuggable. The team's only recourse is to either use the RamseteController example that doesn't use RamseteCommand, or copy-paste the source of RamseteCommand and instrument it with their own logging.

Starlight220, 2021-05-12:
So how would usage/API look like?

calcmogul, 2021-05-12:
Like the "ramsetecontroller" example but with the RamseteController instance moved into the drivetrain subsystem.
```java
  @Override
  public void autonomousInit() {
    // Initialize the timer.
    m_timer = new Timer();
    m_timer.start();

    // Reset the drivetrain's odometry to the starting pose of the trajectory.
    m_drive.resetOdometry(m_trajectory.getInitialPose());
  }

  @Override
  public void autonomousPeriodic() {
    // Update odometry.
    m_drive.updateOdometry();

    if (m_timer.get() < m_trajectory.getTotalTimeSeconds()) {
      // Get the desired pose from the trajectory.
      var desiredPose = m_trajectory.sample(m_timer.get());

      // Get the reference chassis speeds from the Ramsete controller.
      var refChassisSpeeds = m_ramseteController.calculate(m_drive.getPose(), desiredPose);

      // Set the linear and angular speeds.
      m_drive.drive(refChassisSpeeds.vxMetersPerSecond, refChassisSpeeds.omegaRadiansPerSecond);
    } else {
      m_drive.drive(0, 0);
    }
  }
```
The stuff in autonomousInit() would happen in the command start, autonomousPeriodic() would happen in the command's periodic function. The command would end when the time expires.

Starlight220, 2021-05-13:
So we're practically telling teams to implement RamseteCommand themselves? That'll go well...

calcmogul, 2021-05-13:
Well, the current approach is going really poorly because we offer no logging. If it doesn't work out of the box, they have no hope of debugging it. At least by giving an example users can copy-paste, they can add their own logging.

Starlight220, 2021-05-14:
I agree that the following part should be moved, what about the overload where it passes velocity targets to the output lambda?

fixermark, 2023-02-12:
Making use of `RamsetsCommand` this year, I agree that the biggest challenge the team encountered is that it's a black box. Improved logging and additional documentation on how to debug the command would, I think, go quite a ways towards improving its utility.

(Once the team was able to get it working, the results were quite impressive).

closed 2024-06-18 by calcmogul