wpilibsuite/allwpilib #3355: WPILibVersion.java methods used but do not exist at import location.

hallettc, 2021-05-13:
**Describe the bug**
The removal of the WPILibVersion class in edu.wpi.first.wpilibj.util causes error messages and crashing when simulating a robot in Autodesk Synthesis using the auto-generated RobotBase code. 

In RobotBase.java (which can be found here https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/src/main/java/edu/wpi/first/wpilibj/RobotBase.java), take note of line 20:
`import edu.wpi.first.wpilibj.util.WPILibVersion;`
and lines 284-361:
```Java
 private static <T extends RobotBase> void runRobot(Supplier<T> robotSupplier) {
    System.out.println("********** Robot program starting **********");

    T robot;
    try {
      robot = robotSupplier.get();
    } catch (Throwable throwable) {
      Throwable cause = throwable.getCause();
      if (cause != null) {
        throwable = cause;
      }
      String robotName = "Unknown";
      StackTraceElement[] elements = throwable.getStackTrace();
      if (elements.length > 0) {
        robotName = elements[0].getClassName();
      }
      DriverStation.reportError(
          "Unhandled exception instantiating robot " + robotName + " " + throwable.toString(),
          elements);
      DriverStation.reportWarning("Robots should not quit, but yours did!", false);
      DriverStation.reportError("Could not instantiate robot " + robotName + "!", false);
      return;
    }

    m_runMutex.lock();
    m_robotCopy = robot;
    m_runMutex.unlock();

    if (isReal()) {
      try {
        final File file = new File("/tmp/frc_versions/FRC_Lib_Version.ini");

        if (file.exists()) {
          file.delete();
        }

        file.createNewFile();

        try (OutputStream output = Files.newOutputStream(file.toPath())) {
          output.write("Java ".getBytes(StandardCharsets.UTF_8));
          output.write(WPILibVersion.Version.getBytes(StandardCharsets.UTF_8));
        }

      } catch (IOException ex) {
        DriverStation.reportError(
            "Could not write FRC_Lib_Version.ini: " + ex.toString(), ex.getStackTrace());
      }
    }

    boolean errorOnExit = false;
    try {
      robot.startCompetition();
    } catch (Throwable throwable) {
      Throwable cause = throwable.getCause();
      if (cause != null) {
        throwable = cause;
      }
      DriverStation.reportError(
          "Unhandled exception: " + throwable.toString(), throwable.getStackTrace());
      errorOnExit = true;
    } finally {
      m_runMutex.lock();
      boolean suppressExitWarning = m_suppressExitWarning;
      m_runMutex.unlock();
      if (!suppressExitWarning) {
        // startCompetition never returns unless exception occurs....
        DriverStation.reportWarning("Robots should not quit, but yours did!", false);
        if (errorOnExit) {
          DriverStation.reportError(
              "The startCompetition() method (or methods called by it) should have "
                  + "handled the exception above.",
              false);
        } else {
          DriverStation.reportError("Unexpected return from startCompetition() method.", false);
        }
      }
    }
  }

```
However, WPILibVersion.java does not exist at the import location. Therefore, the output cannot be written to output, as the methods at line 322-324 cannot happen:
```Java
        try (OutputStream output = Files.newOutputStream(file.toPath())) {
          output.write("Java ".getBytes(StandardCharsets.UTF_8));
          output.write(WPILibVersion.Version.getBytes(StandardCharsets.UTF_8));
```

When I run my code's Main.java method (autogenerated by WPILib):
```Java
// Copyright (c) FIRST and other WPILib contributors.
// Open Source Software; you can modify and/or share it under the terms of
// the WPILib BSD license file in the root directory of this project.

package frc.robot;

import edu.wpi.first.wpilibj.*;

/**
 * Do NOT add any static variables to this class, or any initialization at all. Unless you know what
 * you are doing, do not modify this file except to change the parameter class to the startRobot
 * call.
 */
public final class Main {
  private Main() {}

  /**
   * Main initialization function. Do not perform any initialization here.
   *
   * <p>If you change your main robot class, change the parameter type.
   */
  public static void main(String... args) {
    RobotBase.startRobot(Robot::new);
  }
}
```

I receive the following log message:

> ********** Robot program starting **********
Error at java.base/java.io.UnixFileSystem.createFileExclusively(Native Method): Could not write FRC_Lib_Version.ini: java.io.IOException: No such file or directory
	at java.base/java.io.UnixFileSystem.createFileExclusively(Native Method)
	at java.base/java.io.File.createNewFile(File.java:1024)
	at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:320)
	at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:407)
	at frc.robot.Main.main(Main.java:23)

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to run any RobotBase code in Autodesk Synthesis using the most up to date version of WPILib and Synthesis. 

**Expected behavior**
Absence of the above log message and java.io.IOException.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Windows 10
 - WPILib Version: v2021.1.2

**Additional context**
While this issue occurred in simulation with Autodesk Synthesis, the error does appear to originate from this particular missing method in WPILib.

prateekma, 2021-05-13:
I'm not sure how you are running your code. If you just copied all the classes from the repo, it won't exist because that class is generated. See: https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/build.gradle#L17

If you are using the release in VS Code, that class should exist. See the sources JAR here: https://frcmaven.wpi.edu/ui/repos/tree/General/release%2Fedu%2Fwpi%2Ffirst%2Fwpilibj%2Fwpilibj-java%2F2021.1.2%2Fwpilibj-java-2021.1.2-sources.jar

ThadHouse, 2021-05-13:
That class also exists in the wpilibj jar. 

https://frcmaven.wpi.edu/artifactory/release/edu/wpi/first/wpilibj/wpilibj-java/2021.3.1/wpilibj-java-2021.3.1.jar

The class file definitely exists in there.

ThadHouse, 2021-05-13:
OH. Youre actually running into a completely different error. The IOException is saying it can't access FRC_Lib_Version.ini.

PeterJohnson, 2021-05-13:
Or since Synthesis is emulating a complete RoboRIO, Synthesis should be providing that file.