wpilibsuite/allwpilib #3357: Unbuilt Source Files

pjreiniger, 2021-05-14:
**Describe the question you have.**
I found some source files that are not build by gradle. I'll gladly help fix, but it should be up to you if it is actually dead code or something that should just be covered by the build system

-  `cscore/java-examples` - The cpp examples (cscore/examples`) are built, but these are not. They currently don't compile, but it is a trivial namespace change
- `ntcore/manualTests` - SUPER old API's, I'm guessing these haven't been used in a while

PeterJohnson, 2021-05-14:
The ntcore C++ manual tests are using recent APIs, but you are correct the Java ones are not.  We should definitely fix the cscore java examples and build them.

pjreiniger, 2021-05-14:
Also `wpiutil/src/test/native/cpp/ManagedStaticTest.cpp` - Builds fine, but fails testing on *nix (https://github.com/pjreiniger/allwpilib/runs/2581731435?check_suite_focus=true)