wpilibsuite/allwpilib #3363: Gradle build fails on -Wmaybe-uninitialized

calcmogul, 2021-05-20:
I'm trying to build allwpilib with GCC 11. The CMake build passes for me, but `./gradlew build` fails with the following (there's other .cpp files too). I'm guessing it's either a bug in Eigen or a false positive, but I've seen address sanitizer warnings like this before from Eigen stuff that suggest it's the former.
```
> Task :wpimath:compileWpimathBaseReleaseStaticLibraryWpimathBaseCpp
In file included from /home/tav/git/wpilib/allwpilib/wpimath/src/main/native/eigeninclude/Eigen/Core:341,
                 from /home/tav/git/wpilib/allwpilib/wpimath/src/main/native/include/frc/StateSpaceUtil.h:12,
                 from /home/tav/git/wpilib/allwpilib/wpimath/src/main/native/cpp/StateSpaceUtil.cpp:5:
/home/tav/git/wpilib/allwpilib/wpimath/src/main/native/eigeninclude/Eigen/src/Core/products/TriangularMatrixVector.h: In function ‘static void Eigen::internal::trmv_selector<Mode, 1>::run(const Lhs&, const Rhs&, Dest&, const typename Dest::Scalar&) [with Lhs = Eigen::Transpose<const Eigen::Block<const Eigen::Block<Eigen::Matrix<double, 2, 2>, -1, -1, false>, -1, -1, false> >; Rhs = Eigen::Transpose<const Eigen::CwiseBinaryOp<Eigen::internal::scalar_product_op<double, double>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Matrix<double, 1, -1, 1, 1, 2> >, const Eigen::Transpose<const Eigen::Block<const Eigen::Block<const Eigen::Block<Eigen::Matrix<double, 2, 2>, -1, -1, false>, -1, 1, true>, -1, 1, false> > > >; Dest = Eigen::Transpose<Eigen::Block<Eigen::Block<Eigen::Matrix<double, -1, -1, 1, -1, -1>, 1, -1, true>, 1, -1, false> >; int Mode = 6]’:
/home/tav/git/wpilib/allwpilib/wpimath/src/main/native/eigeninclude/Eigen/src/Core/products/TriangularMatrixVector.h:332:12: error: ‘result’ may be used uninitialized [-Werror=maybe-uninitialized]
  327 |     internal::triangular_matrix_vector_product
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  328 |       <Index,Mode,
      |       ~~~~~~~~~~~~
  329 |        LhsScalar, LhsBlasTraits::NeedToConjugate,
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  330 |        RhsScalar, RhsBlasTraits::NeedToConjugate,
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  331 |        RowMajor>
      |        ~~~~~~~~~
  332 |       ::run(actualLhs.rows(),actualLhs.cols(),
      |       ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  333 |             actualLhs.data(),actualLhs.outerStride(),
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  334 |             actualRhsPtr,1,
      |             ~~~~~~~~~~~~~~~
  335 |             dest.data(),dest.innerStride(),
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  336 |             actualAlpha);
      |             ~~~~~~~~~~~~
/home/tav/git/wpilib/allwpilib/wpimath/src/main/native/eigeninclude/Eigen/src/Core/products/TriangularMatrixVector.h:105:24: note: by argument 5 of type ‘const double*’ to ‘static void Eigen::internal::triangular_matrix_vector_product<Index, Mode, LhsScalar, ConjLhs, RhsScalar, ConjRhs, 1, Version>::run(Index, Index, const LhsScalar*, Index, const RhsScalar*, Index, Eigen::internal::triangular_matrix_vector_product<Index, Mode, LhsScalar, ConjLhs, RhsScalar, ConjRhs, 1, Version>::ResScalar*, Index, const ResScalar&) [with Index = long int; int Mode = 6; LhsScalar = double; bool ConjLhs = false; RhsScalar = double; bool ConjRhs = false; int Version = 0]’ declared here
  105 | EIGEN_DONT_INLINE void triangular_matrix_vector_product<Index,Mode,LhsScalar,ConjLhs,RhsScalar,ConjRhs,RowMajor,Version>
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

PeterJohnson, 2021-05-20:
This looks like a false positive to me.  Argument 5 is actualRhsPtr, which either points to the passed-in actualRhs or static_rhs, which is copied from actualRhs.  The compiler might not be able to trace the latter assignment to verify it's indeed initialized.

calcmogul, 2021-05-21:
Then we may want to add `-Wno-error=maybe-uninitialized` to our compiler flags.

PR #3378 (2021-05-25): Suppress the -Wmaybe-uninitialized warning in Eigen
GCC 11 emits a false positive when compiling Eigen and breaks the
allwpilib build.

Fixes #3363.
commit 484cf9c0 (2021-05-25): [wpimath] Suppress the -Wmaybe-uninitialized warning in Eigen (#3378)

GCC 11 emits a false positive when compiling Eigen and breaks the
build.

Fixes #3363.
closed 2021-05-25 by PeterJohnson