wpilibsuite/allwpilib #3367: [wpiutil] Potential UB by extending std in wpi::SmallVector

RyanHir, 2021-05-21:
In `wpi/SmallVector.h` there is a instance where std is [being extended](
https://github.com/wpilibsuite/allwpilib/blob/fdbbf118872dd8d4864d5123263b8204d9a2e1b3/wpiutil/src/main/native/include/wpi/SmallVector.h#L931)

While legal for C++17 and below, this is no longer allowed. I understand that C++20 is currently impossible for FRC but I believe this should be removed as std::swap is still usable if the object in question [satisfies the swappable requirment](https://en.cppreference.com/w/cpp/named_req/Swappable).

According to cppreference, ["The expected way to make a program-defined type swappable is to provide a non-member function swap in the same namespace as the type: see Swappable for details."](https://en.cppreference.com/w/cpp/algorithm/swap)

calcmogul, 2021-05-21:
SmallVector came from LLVM's libraries, so it's amusing to see a library written by a compiler vendor being potential UB.

RyanHir, 2021-05-21:
In the current implementation of LLVM, they are still extending std [here](https://github.com/llvm/llvm-project/blob/aa6e7e8da8f5a2706f0b330718df203b3650408e/llvm/include/llvm/ADT/SmallVector.h#L1255)

When used internally by the compiler and toolchain, I would not worry too much. But this version is used my many compilers with different versions so I think this should be looked further if it is necessary.

PeterJohnson, 2021-05-21:
Looks like even in LLVM HEAD they are still overloading std::swap.  https://github.com/llvm/llvm-project/blob/main/llvm/include/llvm/ADT/SmallVector.h#L1255

RyanHir, 2021-05-21:
In that case, I would just consider it sketchy, but if the LLVM team can confirm it working then I guess this is a non-issue.

RyanHir, 2021-05-21:
I think a better fit would be the wpi namespace, or just remove it and let the named requirement work with std::swap https://en.cppreference.com/w/cpp/named_req/Swappable

calcmogul, 2021-05-21:
Maintaining changes on top of a thirdparty library kinda sucks though when you have to resync with upstream.

RyanHir, 2021-05-21:
Would leaving a changelog showing changes from upstream at the begining of the header make it easier for upgrading the library in the future? This would at least document required changed to allow for upgrading in the future...

calcmogul, 2021-05-21:
Something more automatic would be preferred.

RyanHir, 2021-05-21:
If mesa was not a concern, I would propose including the untouched header and aliasing the types to the wpi namespace.

PeterJohnson, 2021-05-29:
It’s not just namespace changes, we have to make substantial edits to cut out irrelevant dependencies.  LLVM upstream has a lot of header coupling that we’ve had to deal with. Also, we are starting to make more incompatible changes like using std::string_view instead of StringRef.  The other thing to note is LLVM did a license change recently to Apache 2.  It’s compatible with our license, but we’ve not taken the leap yet.

RyanHir, 2021-05-31:
I have worked on a little case study to possibly get around the UB of extending std. I think the option of using a wpi::swap may be a good option https://godbolt.org/z/s5eoz9Kns.

msoucy, 2021-06-04:
Unless I'm misreading [cppreference](https://en.cppreference.com/w/cpp/language/extending_std), this is actually legal and not undefined, as it's specializing a standard function for a nonstandard type.

auscompgeek, 2021-06-05:
@msoucy that page says:

> It is undefined behavior to declare a full specialization of any standard library function template. (since C++20)

which I believe matches the scenario here.

msoucy, 2021-06-05:
Conveniently, this doesn't match [the definition of full template specialization](https://en.cppreference.com/w/cpp/language/template_specialization) - since there are still template parameters, it's a partial specialization. So prior to C++20, this appears to be allowed. [The page for swap](https://en.cppreference.com/w/cpp/algorithm/swap) also has specialization allowed (though discouraged) prior to 20.

Related, I hadn't realized that the compilers in use supported C++20 sufficiently enough that they were in use here, so that's a nice surprise.

Long story short: it depends on which standard you're following as to the exact legality, but the recommended approach is what was listed in this issue in either standard.

closed 2021-10-29 by calcmogul