wpilibsuite/allwpilib #3462: Unable to install Wpilib 3.1 for romi

GKCI, 2021-06-26:
*Describe the bug**
A clear and concise description of what the bug is.
The image file is corrupted.

After downloading the WPILibPI_image-v2021.3.1-Romi.zip, I unzip it and the image file appears.  But when either Mounting it or Exploring it is attempted, I get a message that the "image file is corrupted" and therefore cannot be mounted or looked at.  I have tried numerous times downloading the zip file, and four different computers and every time I get the message that the image file is corrupted..

Am unable to mount the image file and therefore cannot get to the installer, and install the software.

What am I doing wrong, or is there another way to get access to the files?

PeterJohnson, 2021-06-26:
The Pi image is designed to be written to a SD card for use in the Raspberry Pi.  For installation of the software suite on the desktop, you want to grab the appropriate installer for your platform from this releases page: https://github.com/wpilibsuite/allwpilib/releases/

GKCI, 2021-06-26:
Thanks for the information. Will do as you have directed.

closed 2021-06-26 by GKCI