wpilibsuite/allwpilib #3468: [Glass] Customizable Number of Data Points in Plot

StarDylan, 2021-07-06:
**Is your feature request related to a problem? Please describe.**
When I use Glass and I pause a plot, I notice that the points start disappearing.

**Describe the solution you'd like**
I would like a customizable plot points for each plot, so that we can make sure that if we need to look at some data, we can see it and keep it, without it disappearing.

I am happy to try and fix it myself and submit a PR. I would also want to add a value in the ini file.

**Describe alternatives you've considered**
I've looked through the code, I didn't see any setting, in fact it is hard coded to be 2000 points max. With a 50hz signal from the robot, that gets exhausted in 40s. Which might not be enough time for all teams.

**Additional context**
If this is a solution that sounds good, I'd be happy to start working on it and submit a PR.

PeterJohnson, 2021-07-06:
Really this is only limited by available memory.  Each plot point takes 16 bytes (X and Y double values), so the 2000 points means 32K of memory (per data series).  This is likely overly conservative given typical computer memory, so bumping that up say 10x is perfectly reasonable, 100x (3.2MB of memory for each series) starts feeling kind of big, but still isn't that unreasonable.

So I'm not sure making it settable is worth the trouble versus just bumping it up across the board.  It's a little harder to make it dynamic than it looks--the current array is not mutex-protected, so writes to it are lock-free; if it was switched to a dynamic size that wouldn't be possible.

StarDylan, 2021-07-06:
Ok, so if we we 100X that constant, that should work, since that gives us ~66 minutes of data points, which is plenty. I can submit a PR if that works.

PeterJohnson, 2021-07-06:
The other thing we need to check is to make sure that many points doesn't kill performance... a quick glance at EmitPlot seems to be that it loops over all points... bumping up the maximum that much is probably a bad idea without doing some pre-filtering during the emit.

StarDylan, 2021-07-06:
So, I did some testing, I changed the max size value to 2 million, to see how much it effects performance on my computer. And compared to before my change, plotting data, it didn't have an effect on CPU or GPU resources. So it seems that it doesn't effect performance quite so much. In addition, the Plotting Library used "ImPlot" says:

> Q: Is ImPlot suitable for plotting large datasets?
> 
> A: Yes, within reason. You can plot tens to hundreds of thousands of points without issue, but don't expect millions to be a buttery smooth experience. That said, you can always downsample extremely large datasets by telling ImPlot to stride your data at larger intervals if needed.

So I think that changing the value to only 200000, should be fine for performance.

Let me know if there is something off with my testing methods, or if you think that pre-filtering is still something that needs to be done.

PeterJohnson, 2021-07-06:
How long did you leave it running, or did you pre-fill the data?  It’s only going to be plotting as much data as is actually in the array.

StarDylan, 2021-07-06:
Hmm, yeah, I forgot about that, I ran it on 50hz for 30 minutes and it took around 5-6x the CPU. I'm not sure what pre-filtering can be done, since all the points should be exposed to the Plotting Library for them to be visible I assume. We could also lower that value to around 30k, that should give us 10 mins of graph time, which should be fine and only use a little extra cpu.

PeterJohnson, 2021-07-07:
The basic concept of a pre-filter depends on the fact the data is time monotonic.  So the approach would be to pull the X (time) limits of the plot (using `ImPlot::GetPlotLimits()`) and get an approximation of what range of the array is visible.  Since the time is monotonic this could be done with a binary search.

PeterJohnson, 2022-02-05:
#3993 increases the plot depth to 20K points.

Starlight220, 2022-05-02:
Fixed by #3993, reopen if not.

closed 2022-05-02 by Starlight220