wpilibsuite/allwpilib #3518: The "angleOffset" in SwerveDrive

mendax1234, 2021-08-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Well , when l use the SwerveBot example on my swerve drive., (l have already adjusted the encoders and motors according to my swerve drive) it seems that the angle of the swerve module is different. l have already searched for some imformation and studied other team's swerve code like 2910. l have found that the "angleOffset" (the angle which needs to be adjusted when the module's wheel is all set to one direction ). The "angleOffset" also needs to be added to the currentangle of each swerve in order to let every swerve module face the same direction during the drive or competition. However , it seems that the SwerveBot example or the SwerveControllerExample doesn't have any part of code to add the "angleOffset".

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

l am not sure of my solution. But l will put my whole code below.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

It seems that there is no alternatives for it.

**Additional context**
Add any other context or screenshots about the feature request here.

The whole code:

calcmogul, 2021-08-17:
If your encoders require an offset, just add it to the encoder value as a constant (assuming these are absolute encoders). You didn't include your source code, but if it's CTRE onboard PID control, we can't really help you. Our docs only cover how to do things with WPILib alone, and leave vendor-specific stuff to the vendors to document themselves (like how to add an offset to a Falcon's internal encoder).

mendax1234, 2021-08-18:
Thanks first, l just only want to come up with a new idea because by using the WPI's official swerve example, our swerve can't work normally. l do use CTRE's encoder , but "angleoffset" idea can be added to swerve example , too because maybe there is not a better way to alter it . 

Well , below l will explain my solution detailedly: 

- First , we need to get the "angleoffset" , since this need CTRE's function , l won't explain.

- After we get the "angleoffset", we should change it into degrees and add it to the getTurningRadians();

```
        m_turningPIDController.calculate(
              getTurningRadians() + mZeroOffset/180 * Math.PI,
              state.angle.getRadians());
```
It seems that this adds the constants to thte encoder value?

However, should l also add the "angleoffset" constants when l use the optimize(). Where should l add it? 

```
        SwerveModuleState.optimize(
              desiredState,
              new Rotation2d(getTurningRadians()+ mZeroOffset/180 * Math.PI));
```

calcmogul, 2021-08-19:
Have `getTurningRadians()` apply the offset, so it's only done in one place for everything.

mendax1234, 2021-08-19:
no, it hasn't . I should plus the offset?

closed 2021-09-20 by calcmogul