wpilibsuite/allwpilib #3536: Make vital functions in compound Subsystem and Command classes non-overridable

Starlight220, 2021-09-03:
Many of the library commands and subsystems such as `PIDSubsystem` and `SequentialCommandGroup` use the interface methods to control their behavior. Everything about this is fine when using commands inline, but problems arise if the team inherits from that command and overrides these methods without calling `super()`. For example, when CSAing I helped a team that had overridden `SequentialCommandGroup.initialize()`, leading to an `IndexOutOfBoundsException` being thrown due to skipped initialization.

Marking these methods as `final` would prevent such cases that are generally hard to debug.

If adding user-defined behavior to these functions is a wanted feature, I suggest using the protected shadow method pattern (ie `_periodic()`- naming TBD) although it raises multiple other questions:
- is the shadow method called before or after the class's internal behavior?
- in `Command.isFinished()`, how do the internal behavior and user-defined behavior interact with each other?
- (slightly derivative of the previous question) if the user-defined `isFinished()` returns `true` and the internal one doesn't (or the other way around), what `interrupted` parameter should be passed to each `end(boolean)` method?

PeterJohnson, 2021-09-11:
I'm thinking some of these "helper" classes are designed for specific use cases and thus should be entirely final (e.g. SequentialCommandGroup)--what's the reason to build a class that derives from it?  In other cases, like PIDSubsystem, I think the worst that could happen is that a user doesn't call super.periodic() and thus the PID doesn't get updated?

Starlight220, 2021-09-11:
I agree with what you said about the Command classes being designed for specific use cases. However, inheriting is the "traditional" way of creating commands, as evidenced by the `HatchbotTraditional` example project. I think the "inline" way is more idiomatic, but removing the inheritance way is a drastic decision I wouldn't make lightly and would need a proper deprecation cycle. In any case, I see that we're in agreement about the Command lifecycle methods being `final`.

Subsystems are a different story. There needs to be a way so both built-in and team functionality is executed, but I don't see a good way of having that (especially not in Java; there might be some hacky magic in C++ that'll accomplish it):
- I share your dislike of shadow methods; they can also become confusing when teams have to use one function when inheriting from `Subsystem(Base)` and another when inheriting from one of the others. 
- The option of `setOnPeriodic(Runnable)` functions has problems similar to the shadow method option: two ways of accomplishing the same thing in possibly intersecting contexts. 

Android `Activity`, `View`, and the like have a similar problem; having to call `super`. Option (2) is used there (for example: `Activity.onTouch` and `Activity.setOnTouchListener`) but there might be some infrastructure backing it that doesn't exist here. In many functions an exception is thrown if `super` is not called, per the [docs](https://developer.android.com/reference/android/app/Activity.html#onCreate(android.os.Bundle)). I don't know what infrastructure backs this though.

It would indeed seem that the cost of not calling `super.periodic` would be the control simply not running; `disabledInit` might have some other effects. In any case, it seems a problem that's relatively simple to find and fix especially if it's extremely documented. Thoughts?

28andrew, 2021-09-11:
> what's the reason to build a class that derives from these?

Not sure if it would've been better if I just made a new InstantCommand with its constructor that takes a runnable but I had a few classes that looked like this:

```java
public class DriveStorePosition extends InstantCommand {
  public DriveStorePosition(DriveTrain driveTrain, int index) {
    super(
        () -> {
          driveTrain.setStoredPosition(index, driveTrain.getPose());
          System.out.println("Stored position: " + driveTrain.getPose());
        },
        driveTrain);
  }
}
```

sciencewhiz, 2021-09-11:
The old command framework used shadow methods. There was additional discussion in #2376

Starlight220, 2021-09-11:
Due to Java's lack of free functions, inheriting commands like @28andrew's example is almost the only way of creating built-in commands with certain commands in a repeatable and self-documenting way. It's also the "traditional" way of creating commands, as I've already said.

<details> <summary>Tangent regarding the example given.</summary>

By the way, this is my preferred way of doing that for single-subsystem commands. It's semantics, but I find this cleaner.
```java
public class DriveTrain extends SubsystemBase {
....
  // CamelCase naming to indicate factory method - your formatting choice
  public Command StorePosition(int index) {
    return new InstantCommand(() -> {
          this.setStoredPosition(index, this.getPose());
          System.out.println("Stored position: " + this.getPose());
        }, this);
  }
 ...
}
```
</details>

The discussion in 2376 shows that it was a conscious design choice that not calling `super.periodic()` etc would silently no-op. [`disabledInit`](3510) might be a bit more problematic since we'd have to define whether `useOutput(0)` is called and what's the responsibility of team code and what is handled by library code -- I'd be happy for opinions in that PR.

---
To sum up, is there an agreement on the following two points?
- Subsystem methods should not be final, but templates, docs, and examples will highlight the need to call `super`.
- Command lifecycle methods will be final, but the classes themselves will be open.

Starlight220, 2021-09-13:
A slight problem: the Gearsbot example overrides commands in a way that this would break, and I'm not sure there is a way to have the same functionality in the "traditional" method once functions are final.

Any ideas?

Starlight220, 2022-09-01:
#4385 handled the four classic command group classes; do we want to mark anything else as `final`?

Starlight220, 2022-10-23:
The main risky overrides were the command groups; that's solved now.

closed 2022-10-23 by Starlight220