wpilibsuite/allwpilib #3542: AnalogEncoderSimTest leaks memory

calcmogul, 2021-09-06:
I got the following warning from GCC 11's address sanitizer.
```
==533712==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 32 byte(s) in 1 object(s) allocated from:
    #0 0x7fbb7da89ca1 in operator new(unsigned long) /build/gcc/src/gcc/libsanitizer/asan/asan_new_delete.cpp:99
    #1 0x7fbb7d434721 in frc::AnalogTrigger::CreateOutput(frc::AnalogTriggerType) const /home/tav/git/wpilib/allwpilib/wpilibc/src/main/native/cpp/AnalogTrigger.cpp:112
    #2 0x7fbb7d40432e in frc::AnalogEncoder::Init() /home/tav/git/wpilib/allwpilib/wpilibc/src/main/native/cpp/AnalogEncoder.cpp:51
    #3 0x7fbb7d403b7a in frc::AnalogEncoder::AnalogEncoder(frc::AnalogInput&) /home/tav/git/wpilib/allwpilib/wpilibc/src/main/native/cpp/AnalogEncoder.cpp:23
    #4 0x56230554346c in AnalogEncoderSimTest_TestBasic_Test::TestBody() /home/tav/git/wpilib/allwpilib/wpilibc/src/test/native/cpp/simulation/AnalogEncoderSimTest.cpp:19
    #5 0x7fbb752a02c5 in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2607
    #6 0x7fbb7528e6e5 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2643
    #7 0x7fbb75226385 in testing::Test::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2682
    #8 0x7fbb75227941 in testing::TestInfo::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2861
    #9 0x7fbb75228aea in testing::TestSuite::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:3015
    #10 0x7fbb7524e516 in testing::internal::UnitTestImpl::RunAllTests() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:5855
    #11 0x7fbb752a2e77 in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2607
    #12 0x7fbb75291196 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2643
    #13 0x7fbb7524af61 in testing::UnitTest::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:5438
    #14 0x562305505046 in RUN_ALL_TESTS() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/include/gtest/gtest.h:2490
    #15 0x562305504f18 in main /home/tav/git/wpilib/allwpilib/wpilibc/src/test/native/cpp/main.cpp:12
    #16 0x7fbb74c3cb24 in __libc_start_main (/usr/lib/libc.so.6+0x27b24)

Direct leak of 32 byte(s) in 1 object(s) allocated from:
    #0 0x7fbb7da89ca1 in operator new(unsigned long) /build/gcc/src/gcc/libsanitizer/asan/asan_new_delete.cpp:99
    #1 0x7fbb7d434721 in frc::AnalogTrigger::CreateOutput(frc::AnalogTriggerType) const /home/tav/git/wpilib/allwpilib/wpilibc/src/main/native/cpp/AnalogTrigger.cpp:112
    #2 0x7fbb7d404247 in frc::AnalogEncoder::Init() /home/tav/git/wpilib/allwpilib/wpilibc/src/main/native/cpp/AnalogEncoder.cpp:49
    #3 0x7fbb7d403b7a in frc::AnalogEncoder::AnalogEncoder(frc::AnalogInput&) /home/tav/git/wpilib/allwpilib/wpilibc/src/main/native/cpp/AnalogEncoder.cpp:23
    #4 0x56230554346c in AnalogEncoderSimTest_TestBasic_Test::TestBody() /home/tav/git/wpilib/allwpilib/wpilibc/src/test/native/cpp/simulation/AnalogEncoderSimTest.cpp:19
    #5 0x7fbb752a02c5 in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2607
    #6 0x7fbb7528e6e5 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2643
    #7 0x7fbb75226385 in testing::Test::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2682
    #8 0x7fbb75227941 in testing::TestInfo::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2861
    #9 0x7fbb75228aea in testing::TestSuite::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:3015
    #10 0x7fbb7524e516 in testing::internal::UnitTestImpl::RunAllTests() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:5855
    #11 0x7fbb752a2e77 in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2607
    #12 0x7fbb75291196 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2643
    #13 0x7fbb7524af61 in testing::UnitTest::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:5438
    #14 0x562305505046 in RUN_ALL_TESTS() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/include/gtest/gtest.h:2490
    #15 0x562305504f18 in main /home/tav/git/wpilib/allwpilib/wpilibc/src/test/native/cpp/main.cpp:12
    #16 0x7fbb74c3cb24 in __libc_start_main (/usr/lib/libc.so.6+0x27b24)

SUMMARY: AddressSanitizer: 64 byte(s) leaked in 2 allocation(s).
```

PeterJohnson, 2021-09-09:
This appears to be due to an incorrect use of NullDeleter here: https://github.com/wpilibsuite/allwpilib/blob/main/wpilibc/src/main/native/cpp/AnalogTrigger.cpp#L113

calcmogul, 2021-09-09:
Yep, that fixed it.

PR #3554 (2021-09-10): Fix AnalogTriggerOutput memory leak reported by asan
Fixes #3542.
commit 3d9a4d58 (2021-09-10): [wpilibc] Fix AnalogTriggerOutput memory leak reported by asan (#3554)

Fixes #3542.
closed 2021-09-10 by PeterJohnson