wpilibsuite/allwpilib #3544: libuv tests leak memory

calcmogul, 2021-09-06:
I got the following warning from GCC 11's address sanitizer.
```
==533769==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 8192 byte(s) in 1 object(s) allocated from:
    #0 0x7f9e8564ee21 in operator new[](unsigned long) /build/gcc/src/gcc/libsanitizer/asan/asan_new_delete.cpp:102
    #1 0x55bebdc7f9b0 in wpi::uv::Buffer::Allocate(unsigned long) /home/tav/git/wpilib/allwpilib/wpiutil/src/main/native/include/wpi/uv/Buffer.h:52
    #2 0x55bebe747545 in wpi::uv::SimpleBufferPool<4ul>::Allocate() /home/tav/git/wpilib/allwpilib/wpiutil/src/main/native/include/wpi/uv/Buffer.h:116
    #3 0x55bebe744c53 in wpi::uv::UvSimpleBufferPool_ConstructSize_Test::TestBody() /home/tav/git/wpilib/allwpilib/wpiutil/src/test/native/cpp/uv/UvBufferTest.cpp:19
    #4 0x7f9e84e542c5 in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2607
    #5 0x7f9e84e426e5 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2643
    #6 0x7f9e84dda385 in testing::Test::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2682
    #7 0x7f9e84ddb941 in testing::TestInfo::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2861
    #8 0x7f9e84ddcaea in testing::TestSuite::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:3015
    #9 0x7f9e84e02516 in testing::internal::UnitTestImpl::RunAllTests() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:5855
    #10 0x7f9e84e56e77 in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2607
    #11 0x7f9e84e45196 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2643
    #12 0x7f9e84dfef61 in testing::UnitTest::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:5438
    #13 0x55bebe5cc472 in RUN_ALL_TESTS() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/include/gtest/gtest.h:2490
    #14 0x55bebe5cc343 in main /home/tav/git/wpilib/allwpilib/wpiutil/src/test/native/cpp/main.cpp:9
    #15 0x7f9e847f0b24 in __libc_start_main (/usr/lib/libc.so.6+0x27b24)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f9e8564ee21 in operator new[](unsigned long) /build/gcc/src/gcc/libsanitizer/asan/asan_new_delete.cpp:102
    #1 0x55bebdc7f9b0 in wpi::uv::Buffer::Allocate(unsigned long) /home/tav/git/wpilib/allwpilib/wpiutil/src/main/native/include/wpi/uv/Buffer.h:52
    #2 0x55bebe747545 in wpi::uv::SimpleBufferPool<4ul>::Allocate() /home/tav/git/wpilib/allwpilib/wpiutil/src/main/native/include/wpi/uv/Buffer.h:116
    #3 0x55bebe7447a1 in wpi::uv::UvSimpleBufferPool_ConstructDefault_Test::TestBody() /home/tav/git/wpilib/allwpilib/wpiutil/src/test/native/cpp/uv/UvBufferTest.cpp:13
    #4 0x7f9e84e542c5 in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2607
    #5 0x7f9e84e426e5 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2643
    #6 0x7f9e84dda385 in testing::Test::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2682
    #7 0x7f9e84ddb941 in testing::TestInfo::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2861
    #8 0x7f9e84ddcaea in testing::TestSuite::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:3015
    #9 0x7f9e84e02516 in testing::internal::UnitTestImpl::RunAllTests() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:5855
    #10 0x7f9e84e56e77 in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2607
    #11 0x7f9e84e45196 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2643
    #12 0x7f9e84dfef61 in testing::UnitTest::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:5438
    #13 0x55bebe5cc472 in RUN_ALL_TESTS() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/include/gtest/gtest.h:2490
    #14 0x55bebe5cc343 in main /home/tav/git/wpilib/allwpilib/wpiutil/src/test/native/cpp/main.cpp:9
    #15 0x7f9e847f0b24 in __libc_start_main (/usr/lib/libc.so.6+0x27b24)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f9e8564ee21 in operator new[](unsigned long) /build/gcc/src/gcc/libsanitizer/asan/asan_new_delete.cpp:102
    #1 0x55bebdc7f9b0 in wpi::uv::Buffer::Allocate(unsigned long) /home/tav/git/wpilib/allwpilib/wpiutil/src/main/native/include/wpi/uv/Buffer.h:52
    #2 0x55bebe747545 in wpi::uv::SimpleBufferPool<4ul>::Allocate() /home/tav/git/wpilib/allwpilib/wpiutil/src/main/native/include/wpi/uv/Buffer.h:116
    #3 0x55bebe74517d in wpi::uv::UvSimpleBufferPool_ReleaseReuse_Test::TestBody() /home/tav/git/wpilib/allwpilib/wpiutil/src/test/native/cpp/uv/UvBufferTest.cpp:25
    #4 0x7f9e84e542c5 in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2607
    #5 0x7f9e84e426e5 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2643
    #6 0x7f9e84dda385 in testing::Test::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2682
    #7 0x7f9e84ddb941 in testing::TestInfo::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2861
    #8 0x7f9e84ddcaea in testing::TestSuite::Run() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:3015
    #9 0x7f9e84e02516 in testing::internal::UnitTestImpl::RunAllTests() /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:5855
    #10 0x7f9e84e56e77 in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2607
    #11 0x7f9e84e45196 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-asan/googletest/googletest-src/googletest/src/gtest.cc:2643
[... cut ...]
    #12 0x7f9e8520ceb1 in _M_invoke<0> /usr/include/c++/11.1.0/bits/std_thread.h:253
    #13 0x7f9e8520ce85 in operator() /usr/include/c++/11.1.0/bits/std_thread.h:260
    #14 0x7f9e8520ce69 in _M_run /usr/include/c++/11.1.0/bits/std_thread.h:211
    #15 0x7f9e84bea3c3 in execute_native_thread_routine /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:82

SUMMARY: AddressSanitizer: 22664 byte(s) leaked in 20 allocation(s).
```
I ran wpiutil_test from a CMake build of https://github.com/calcmogul/allwpilib/tree/sanitizers with the Asan build type.

PeterJohnson, 2021-09-09:
The first few are bugs in the unit tests (the allocated buffers must be released back into the pool explicitly).  Similarly raw_uv_stream_test.cpp is missing explicit buffer releases.  Not sure about the thread ones yet.

calcmogul, 2021-09-09:
That fixes the buffer tests specifically, but it doesn't address the memory leaks in the web socket stuff.

PeterJohnson, 2021-09-09:
Do you have traces for the web socket ones?  The rest of the ones above all seem to be worker thread tests?

calcmogul, 2021-09-09:
Hm, I think those were only from when I tried RAII-ifying uv::Buffer. The ones for main only talk about the buffer tests and worker thread tests.

closed 2021-09-10 by PeterJohnson