wpilibsuite/allwpilib #3546: Double-lock of mutex in cscore HTTP camera test

calcmogul, 2021-09-06:
I got the following warning from GCC 11's thread sanitizer.
```
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from CameraSourceTest
[ RUN      ] CameraSourceTest.HTTPCamera
==================
WARNING: ThreadSanitizer: double lock of a mutex (pid=534353)
    #0 pthread_mutex_lock /build/gcc/src/gcc/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:4250 (libtsan.so.0+0x54b6a)
    #1 __gthread_mutex_lock /usr/include/c++/11.1.0/x86_64-pc-linux-gnu/bits/gthr-default.h:749 (libcscore.so+0x12a094)
    #2 std::mutex::lock() /usr/include/c++/11.1.0/bits/std_mutex.h:100 (libcscore.so+0x12a118)
    #3 std::unique_lock<std::mutex>::lock() /usr/include/c++/11.1.0/bits/unique_lock.h:139 (libcscore.so+0x151be7)
    #4 std::unique_lock<std::mutex>::unique_lock(std::mutex&) /usr/include/c++/11.1.0/bits/unique_lock.h:69 (libcscore.so+0x14e44a)
    #5 cs::HttpCameraImpl::StreamThreadMain() /home/tav/git/wpilib/allwpilib/cscore/src/main/native/cpp/HttpCameraImpl.cpp:108 (libcscore.so+0x14411b)
    #6 void std::__invoke_impl<void, void (cs::HttpCameraImpl::*)(), cs::HttpCameraImpl*>(std::__invoke_memfun_deref, void (cs::HttpCameraImpl::*&&)(), cs::HttpCameraImpl*&&) <null> (libcscore.so+0x15d127)
    #7 std::__invoke_result<void (cs::HttpCameraImpl::*)(), cs::HttpCameraImpl*>::type std::__invoke<void (cs::HttpCameraImpl::*)(), cs::HttpCameraImpl*>(void (cs::HttpCameraImpl::*&&)(), cs::HttpCameraImpl*&&) <null> (libcscore.so+0x15cfea)
    #8 void std::thread::_Invoker<std::tuple<void (cs::HttpCameraImpl::*)(), cs::HttpCameraImpl*> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>) <null> (libcscore.so+0x15cefa)
    #9 std::thread::_Invoker<std::tuple<void (cs::HttpCameraImpl::*)(), cs::HttpCameraImpl*> >::operator()() <null> (libcscore.so+0x15cdce)
    #10 std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (cs::HttpCameraImpl::*)(), cs::HttpCameraImpl*> > >::_M_run() <null> (libcscore.so+0x15ccac)
    #11 execute_native_thread_routine /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:82 (libstdc++.so.6+0xd33c3)

  Location is heap block of size 760 at 0x7b5800000f00 allocated by main thread:
    #0 operator new(unsigned long) /build/gcc/src/gcc/libsanitizer/tsan/tsan_new_delete.cpp:64 (libtsan.so.0+0x91824)
    #1 __gnu_cxx::new_allocator<std::_Sp_counted_ptr_inplace<cs::HttpCameraImpl, std::allocator<cs::HttpCameraImpl>, (__gnu_cxx::_Lock_policy)2> >::allocate(unsigned long, void const*) /usr/include/c++/11.1.0/ext/new_allocator.h:121 (libcscore.so+0x15bf37)
    #2 std::allocator_traits<std::allocator<std::_Sp_counted_ptr_inplace<cs::HttpCameraImpl, std::allocator<cs::HttpCameraImpl>, (__gnu_cxx::_Lock_policy)2> > >::allocate(std::allocator<std::_Sp_counted_ptr_inplace<cs::HttpCameraImpl, std::allocator<cs::HttpCameraImpl>, (__gnu_cxx::_Lock_policy)2> >&, unsigned long) /usr/include/c++/11.1.0/bits/alloc_traits.h:460 (libcscore.so+0x15b8eb)
    #3 std::__allocated_ptr<std::allocator<std::_Sp_counted_ptr_inplace<cs::HttpCameraImpl, std::allocator<cs::HttpCameraImpl>, (__gnu_cxx::_Lock_policy)2> > > std::__allocate_guarded<std::allocator<std::_Sp_counted_ptr_inplace<cs::HttpCameraImpl, std::allocator<cs::HttpCameraImpl>, (__gnu_cxx::_Lock_policy)2> > >(std::allocator<std::_Sp_counted_ptr_inplace<cs::HttpCameraImpl, std::allocator<cs::HttpCameraImpl>, (__gnu_cxx::_Lock_policy)2> >&) /usr/include/c++/11.1.0/bits/allocated_ptr.h:97 (libcscore.so+0x15afd7)
    #4 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<cs::HttpCameraImpl, std::allocator<cs::HttpCameraImpl>, std::basic_string_view<char, std::char_traits<char> >&, CS_HttpCameraKind&, wpi::Logger&, cs::Notifier&, cs::Telemetry&>(cs::HttpCameraImpl*&, std::_Sp_alloc_shared_tag<std::allocator<cs::HttpCameraImpl> >, std::basic_string_view<char, std::char_traits<char> >&, CS_HttpCameraKind&, wpi::Logger&, cs::Notifier&, cs::Telemetry&) /usr/include/c++/11.1.0/bits/shared_ptr_base.h:648 (libcscore.so+0x15a3f5)
    #5 std::__shared_ptr<cs::HttpCameraImpl, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<cs::HttpCameraImpl>, std::basic_string_view<char, std::char_traits<char> >&, CS_HttpCameraKind&, wpi::Logger&, cs::Notifier&, cs::Telemetry&>(std::_Sp_alloc_shared_tag<std::allocator<cs::HttpCameraImpl> >, std::basic_string_view<char, std::char_traits<char> >&, CS_HttpCameraKind&, wpi::Logger&, cs::Notifier&, cs::Telemetry&) <null> (libcscore.so+0x15951b)
    #6 std::shared_ptr<cs::HttpCameraImpl>::shared_ptr<std::allocator<cs::HttpCameraImpl>, std::basic_string_view<char, std::char_traits<char> >&, CS_HttpCameraKind&, wpi::Logger&, cs::Notifier&, cs::Telemetry&>(std::_Sp_alloc_shared_tag<std::allocator<cs::HttpCameraImpl> >, std::basic_string_view<char, std::char_traits<char> >&, CS_HttpCameraKind&, wpi::Logger&, cs::Notifier&, cs::Telemetry&) <null> (libcscore.so+0x157c18)
    #7 std::shared_ptr<cs::HttpCameraImpl> std::allocate_shared<cs::HttpCameraImpl, std::allocator<cs::HttpCameraImpl>, std::basic_string_view<char, std::char_traits<char> >&, CS_HttpCameraKind&, wpi::Logger&, cs::Notifier&, cs::Telemetry&>(std::allocator<cs::HttpCameraImpl> const&, std::basic_string_view<char, std::char_traits<char> >&, CS_HttpCameraKind&, wpi::Logger&, cs::Notifier&, cs::Telemetry&) /usr/include/c++/11.1.0/bits/shared_ptr.h:861 (libcscore.so+0x154f94)
    #8 std::shared_ptr<cs::HttpCameraImpl> std::make_shared<cs::HttpCameraImpl, std::basic_string_view<char, std::char_traits<char> >&, CS_HttpCameraKind&, wpi::Logger&, cs::Notifier&, cs::Telemetry&>(std::basic_string_view<char, std::char_traits<char> >&, CS_HttpCameraKind&, wpi::Logger&, cs::Notifier&, cs::Telemetry&) /usr/include/c++/11.1.0/bits/shared_ptr.h:877 (libcscore.so+0x15088b)
    #9 cs::CreateHttpCamera(std::basic_string_view<char, std::char_traits<char> >, std::basic_string_view<char, std::char_traits<char> >, CS_HttpCameraKind, int*) /home/tav/git/wpilib/allwpilib/cscore/src/main/native/cpp/HttpCameraImpl.cpp:570 (libcscore.so+0x1477de)
    #10 cs::HttpCamera::HttpCamera(std::basic_string_view<char, std::char_traits<char> >, char const*, cs::HttpCamera::HttpCameraKind) /home/tav/git/wpilib/allwpilib/cscore/src/main/native/include/cscore_oo.inc:295 (cscore_test+0x48f9)
    #11 cs::CameraSourceTest_HTTPCamera_Test::TestBody() /home/tav/git/wpilib/allwpilib/cscore/src/test/native/cpp/CameraSourceTest.cpp:16 (cscore_test+0x4478)
    #12 void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-tsan/googletest/googletest-src/googletest/src/gtest.cc:2607 (libgtest.so.1.11.0+0xd83f4)
    #13 void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-tsan/googletest/googletest-src/googletest/src/gtest.cc:2643 (libgtest.so.1.11.0+0xcc73e)
    #14 testing::Test::Run() /home/tav/git/wpilib/allwpilib/build-cmake-tsan/googletest/googletest-src/googletest/src/gtest.cc:2682 (libgtest.so.1.11.0+0x95f67)
    #15 testing::TestInfo::Run() /home/tav/git/wpilib/allwpilib/build-cmake-tsan/googletest/googletest-src/googletest/src/gtest.cc:2861 (libgtest.so.1.11.0+0x96c4d)
    #16 testing::TestSuite::Run() /home/tav/git/wpilib/allwpilib/build-cmake-tsan/googletest/googletest-src/googletest/src/gtest.cc:3015 (libgtest.so.1.11.0+0x978d2)
    #17 testing::internal::UnitTestImpl::RunAllTests() /home/tav/git/wpilib/allwpilib/build-cmake-tsan/googletest/googletest-src/googletest/src/gtest.cc:5855 (libgtest.so.1.11.0+0xa91cb)
    #18 bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-tsan/googletest/googletest-src/googletest/src/gtest.cc:2607 (libgtest.so.1.11.0+0xd9c6d)
    #19 bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/tav/git/wpilib/allwpilib/build-cmake-tsan/googletest/googletest-src/googletest/src/gtest.cc:2643 (libgtest.so.1.11.0+0xce21c)
[... cut ...]
SUMMARY: ThreadSanitizer: double lock of a mutex /usr/include/c++/11.1.0/x86_64-pc-linux-gnu/bits/gthr-default.h:749 in __gthread_mutex_lock
==================
[       OK ] CameraSourceTest.HTTPCamera (1005 ms)
[----------] 1 test from CameraSourceTest (1005 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (1005 ms total)
[  PASSED  ] 1 test.
ThreadSanitizer: reported 1 warnings
```
I ran cscore_test from a CMake build of https://github.com/calcmogul/allwpilib/tree/sanitizers with the Tsan build type.

PeterJohnson, 2021-09-09:
What might fix this is calling `cs::Shutdown();`

calcmogul, 2021-09-10:
Yep, that did it.

PR #3557 (2021-09-10): [cscore] Fix cscore test mutex use-after-free caught by tsan
Fixes #3546.
commit 2dd5701a (2021-09-10): [cscore] Fix mutex use-after-free in cscore test (#3557)

Fixes #3546.
closed 2021-09-10 by PeterJohnson