wpilibsuite/allwpilib #3560: frc::filesystem doxygen has incorrect page name

dawonn, 2021-09-12:
https://first.wpi.edu/FRC/roborio/release/docs/cpp/namespacefrc_1_1filesystem.html

AustinShalit, 2021-09-12:
I wonder if this is related to #3392?

AustinShalit, 2021-09-12:
Note that Doxygen is producing the correct output. Filesystem is only a namespace and not a class - https://github.com/wpilibsuite/allwpilib/blob/main/wpilibc/src/main/native/include/frc/Filesystem.h

sciencewhiz, 2021-09-12:
In the 2022 development docs it has its own entry. 

[image]

FYI, you linked the 2020 docs, 2021 docs are here: https://first.wpi.edu/wpilib/allwpilib/docs/release/cpp/index.html

dawonn, 2021-09-12:
double thanks. it was a bit difficult to discover but I see it now.

PeterJohnson, 2021-09-16:
The fact this is done with a namespace instead of a class with static functions is fairly common in broader C++ but is unusual for WPILibC.  For consistency with other classes (e.g. SmartDashboard) and Java, should this be changed to be a normal class with static functions?

sciencewhiz, 2022-01-23:
Closing since 2022 released and it has it's own entry

closed 2022-01-23 by sciencewhiz