wpilibsuite/allwpilib #3567: error writing 1 byte into a region of size 0 when compiling with cmake

WilliamBehrens, 2021-09-14:
**Describe the bug**
Building with flags
-DWITH_GUI=ON
-DWITH_JAVA=OFF
-DWITH_TESTS=OFF
-DWITH_CSCORE=ON
-DWITH_WPILIB=ON
-DWITH_WPIMATH=ON
-DWITH_SHARED_LIBS=OFF
-DWITH_SIMULATION_MODULES=ON

causes the error 
```
simulation/halsim_ds_socket/src/main/native/cpp/DSCommPacket.cpp:244:36: error: writing 1 byte into a region of size 0 -Werror=stringop-overflow=]
  244 |     packet.descriptor.axisTypes[i] = data[1 + i];
```

**To Reproduce**
Steps to reproduce the behavior:
1. Generate cmake with the above mentioned flags
2. build project with cmake --build

**Expected behavior**
Libraries to be output with no errors

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Void Linux  glibc x86_64
 - GCC version: 10.2.1

calcmogul, 2021-09-14:
See if this fixes it. It's based on the bounds checking in the code right above that.
```diff
diff --git a/simulation/halsim_ds_socket/src/main/native/cpp/DSCommPacket.cpp b/simulation/halsim_ds_socket/src/main/native/cpp/DSCommPacket.cpp
index 9088bd7ee..226978056 100644
--- a/simulation/halsim_ds_socket/src/main/native/cpp/DSCommPacket.cpp
+++ b/simulation/halsim_ds_socket/src/main/native/cpp/DSCommPacket.cpp
@@ -237,7 +237,8 @@ void DSCommPacket::ReadJoystickDescriptionTag(wpi::span<const uint8_t> data) {
   }
   data = data.subspan(4 + nameLength);
   packet.descriptor.name[nameLength] = '\0';
-  int axesCount = data[0];
+  int axesCount =
+      std::min<size_t>(data[0], (sizeof(packet.descriptor.axisCount) - 1));
   packet.descriptor.axisCount = axesCount;
   for (int i = 0; i < axesCount; i++) {
     packet.descriptor.axisTypes[i] = data[1 + i];
```

WilliamBehrens, 2021-09-15:
Yep that fixed it, I was building off the latest release and not master if that matters at all aswell

PR #3568 (2021-09-15): Fix halsim_ds_socket stringop overflow warning from GCC 10
Fixes #3567.
commit 65797caa (2021-09-15): [sim] Fix halsim_ds_socket stringop overflow warning from GCC 10 (#3568)

Fixes #3567.
closed 2021-09-15 by PeterJohnson