wpilibsuite/allwpilib #3604: CMake does not currently allow for building wpilibj and wpilibc seperatly

WilliamBehrens, 2021-09-27:
**Describe the bug**
CMake does not currently allow for building wpilibj and wpilibc seperatly. When using the CMake build support with wpilib say I run `cmake ..` just as I'd expect I'd get all build options wpilibj/c, wpiutil, etc. But say if I used `cmake .. -DWITH_JAVA=OFF` if I were to only want to build wpilibc and it's deps it doesn't work. It'll only build the extra libs (wpiutil, ntcore, etc) but wpilibc will be missing and doesn't get compiled.

**To Reproduce**
Steps to reproduce the behavior:
1. `mkdir build/ && cd build/`
2. `cmake .. -DWITH_JAVA=OFF`

**Expected behavior**
wpilibc to be built without and wpilibj requirements

**Desktop (please complete the following information):**
 - Void Linux
 - Java version: 11.0.12
 - C++ version: g++ 10.2.1

**Additional context**
cmake version: 3.21.1

prateekma, 2021-09-28:
I can compile `libwpilibc.dylib` with `-DWITH_JAVA=OFF`. It's in `lib/` inside the CMake build directory.

ThadHouse, 2021-09-28:
I just double checked this as well, and wpilibc is being built. I wonder if something is weird on your platform.

As a side effect, I double checked, and its actually possible for us to remove the cscore dependency on wpilibc. I'm submitting a PR to make that possible.

ThadHouse, 2021-09-28:
https://github.com/wpilibsuite/allwpilib/pull/3605

Here's the PR to allow wpilibc without cmake. At least locally, like Prateek this works fine without Java.

WilliamBehrens, 2021-09-29:
huh weird, I've been having this issue for a few days but I just tried again with the same commit and like everything worked, strange, also being able to remove the opencv dependancy would be great if it's not actually required

calcmogul, 2021-09-29:
Closed by #3605.

closed 2021-09-29 by calcmogul