wpilibsuite/allwpilib #3606: Servos not viewable in simulation GUI

PeterJohnson, 2021-09-28:
Servo objects set HAL PWM position, but the simulation GUI only shows PWM speed, with no way to view position.  The net result of this is that servos don't show up in the GUI at all.  A couple of options to fix this:
- Similar to AddressableLED, provide a way to set a PWM output to "position" mode, then use that info in the simulation GUI to show position instead of speed.
- Unify the PWM speed and position values in simulation similar to how they behave on real hardware (in the Athena HAL, both setPosition and setSpeed ultimately result in a raw PWM set).

sciencewhiz, 2021-09-28:
I think I prefer #2.

PeterJohnson, 2021-09-28:
Note one downside of option 2 is that we'll still display the output as a -1 to +1 range, even though set position takes a 0-1 range... the 0-1 range will be mapped to -1 to 1 (0 -> -1, 0.5 -> 0, 1 -> 1).

sciencewhiz, 2021-12-19:
Is this also the root cause of only Servo setAngle working on Romi, or is that separate?

https://www.chiefdelphi.com/t/romi-arm/390825/17

PeterJohnson, 2021-12-19:
No; both set() and setAngle() call setPosition().

PR #5277 (2023-07-10): Unify PWM simulation Speed, Position, and Raw
Setting one will set the others, like it does in real hardware 
Add tests for boundary conditions and conversions
Update PWM sendable implementation to include all forms 
Fixes #5264
Fixes #3606
commit f8e74e2f (2023-07-10): [hal] Unify PWM simulation Speed, Position, and Raw (#5277)

Setting one will set the others, like it does in real hardware.
Add tests for boundary conditions and conversions.
Update PWM sendable implementation to include all forms.
Fixes #5264
Fixes #3606
closed 2023-07-10 by PeterJohnson