wpilibsuite/allwpilib #3718: Java Robot Simulation GUI not working on MacOS

3LucasZ, 2021-11-13:
**Describe the bug**
Running simulation GUI on Mac OS (gradle task `simulateJava`) (tried with multiple projects) creates a java process but gets stuck Initializing.

**Log Output:**
HAL Extensions: Attempting to load: libhalsim_gui
Simulator GUI Initializing.

**To Reproduce**
Steps to reproduce the behavior:
1. on mac, download wpilib
2. clone gearsbot repo
3. setup and run gradle task `simulateJava`
4. the simulation GUI does not show up

**Expected behavior**
A clear and concise description of what you expected to happen.
I expected Simulation GUI to show up, with the project being run and simulated on it.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - WPILib Version: 2021.3.1
 - OS: MacOS 11.4
 - Java version 11.0.8+10
 - C++ version N/A

**Additional context**
Add any other context about the problem here.

PeterJohnson, 2021-11-13:
Can you try creating an example project within WPILib instead of using Gearsbot?

Also, does Glass (until tools) run successfully?

3LucasZ, 2021-11-13:
An example project within vscode does work.
Although the robot simulation GUI does run successfully, it doesn't work on an external IDE such as IntelliJ.

closed 2021-11-13 by 3LucasZ

PeterJohnson, 2021-11-13:
Are you still running gradlew simulateJava in that case?