wpilibsuite/allwpilib #3725: Can not build RomiReference project in 2022.1.1b2

swalberg, 2021-11-19:
The RomiReference Example in 2022.1.1b2 fails to build or connect to the robot.

Steps to reproduce the behavior:
0. On a Macbook Pro running Big Sur
1. Install  WPILib_macOS-2022.1.1-beta-2.dmg and WPILib-VSCode-1.62.0.zip
2. Generate a new WPILib Project, choosing Example, Java, and RomiReference.
3. Try to build, observe the following error:
```
> Executing task: ./gradlew build   -Dorg.gradle.java.home="/Users/sean/wpilib/2022/jdk" <

FAILURE: Build failed with an exception.

* Where:
Build file '/Users/sean/FRCProjects/romi5/build.gradle' line: 20

* What went wrong:
A problem occurred evaluating root project 'romi5'.
> Could not find method roborioDebug() for arguments [[provider(?), provider(?), provider(?), provider(?), provider(?), provider(?)]] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
The terminal process "bash '-c', './gradlew build   -Dorg.gradle.java.home="/Users/sean/wpilib/2022/jdk"'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
4. Comment out the following lines in `build.gradle`:
```
    roborioDebug wpi.java.deps.wpilibJniDebug(wpi.platforms.roborio)
    roborioDebug wpi.java.vendor.jniDebug(wpi.platforms.roborio)

    roborioRelease wpi.java.deps.wpilibJniRelease(wpi.platforms.roborio)
    roborioRelease wpi.java.vendor.jniRelease(wpi.platforms.roborio)
```
5. Note successful build, start the simulator
6. Observe that the websocket never connects. The terminal output shows `Will attempt to connect to ws://localhost:3300/wpilibws` despite the presence of `wpi.sim.envVar("HALSIMWS_HOST", "10.0.0.2")` in `build.gradle.

My expectation is that I can build an example Romi project and drive my bot.
* Generating a tank drive example builds, and `build.gradle` looks different.
* This procedure works on the 2021.3.1 release

 - WPILib Version: 2022.1.1-beta-2
 - OS: OSX 11.6.1
 - Java version: `openjdk version "11.0.12" 2021-07-20`
 - C++ version [e.g. 17] : n/a

**Additional context**
I removed the contents of ~/wpilib before trying this.

ThadHouse, 2021-11-19:
Also, how are you starting the simulator. Are you starting it through VS Code or from the command line?

swalberg, 2021-11-19:
Ah, thanks. Starting the simulator with the F5 key.

swalberg, 2021-11-19:
I see in the history of the build.gradle template that the way to set the envvar has changed over time. For reference, here is the `build.gradle` that I'm using

```Groovy
plugins {
    id "java"
    id "edu.wpi.first.GradleRIO" version "2022.1.1-beta-2"
}

sourceCompatibility = JavaVersion.VERSION_11
targetCompatibility = JavaVersion.VERSION_11

def ROBOT_MAIN_CLASS = "frc.robot.Main"

// Set this to true to enable desktop support.
def includeDesktopSupport = true

// Defining my dependencies. In this case, WPILib (+ friends), and vendor libraries.
// Also defines JUnit 4.
dependencies {
    implementation wpi.java.deps.wpilib()
    implementation wpi.java.vendor.java()

  /*
    roborioDebug wpi.java.deps.wpilibJniDebug(wpi.platforms.roborio)
    roborioDebug wpi.java.vendor.jniDebug(wpi.platforms.roborio)

    roborioRelease wpi.java.deps.wpilibJniRelease(wpi.platforms.roborio)
    roborioRelease wpi.java.vendor.jniRelease(wpi.platforms.roborio)
*/
    nativeDebug wpi.java.deps.wpilibJniDebug(wpi.platforms.desktop)
    nativeDebug wpi.java.vendor.jniDebug(wpi.platforms.desktop)
    simulationDebug wpi.sim.enableDebug()

    nativeRelease wpi.java.deps.wpilibJniRelease(wpi.platforms.desktop)
    nativeRelease wpi.java.vendor.jniRelease(wpi.platforms.desktop)
    simulationRelease wpi.sim.enableRelease()

    testImplementation 'junit:junit:4.12'
}

// Simulation configuration (e.g. environment variables).
wpi.sim.addGui().defaultEnabled = true
wpi.sim.addDriverstation()

//Sets the websocket client remote host.
wpi.sim.envVar("HALSIMWS_HOST", "10.0.0.2")
wpi.sim.addWebsocketsServer().defaultEnabled = true
wpi.sim.addWebsocketsClient().defaultEnabled = true

// Setting up my Jar File. In this case, adding all libraries into the main jar ('fat jar')
// in order to make them all available at runtime. Also adding the manifest so WPILib
// knows where to look for our Robot Class.
jar {
    from { configurations.runtimeClasspath.collect { it.isDirectory() ? it : zipTree(it) } }
    manifest edu.wpi.first.gradlerio.GradleRIOPlugin.javaManifest(ROBOT_MAIN_CLASS)
    duplicatesStrategy = DuplicatesStrategy.INCLUDE
}

wpi.java.configureExecutableTasks(jar)
wpi.java.configureTestTasks(test)

```

ThadHouse, 2021-11-19:
GradleRIO got a complete rewrite in 2022, so the way things like that worked changed. I just found the bug, its actually on the VS Code side. Pushed the fix so it will be fixed for next beta as well. A workaround for now is to launch the simulate task not through VS Code's debug engine. The best way to do this is there is a command in the VS Code Command Palette called `WPILib: Run a command in Gradle`. Run that, and in the prompt enter `simulateJava`. That will run the simulator inside of gradle which doesn't have the issue, instead of inside VS Code.

swalberg, 2021-11-19:
Thanks for that update. I've tried that and it keeps hanging at 75%. That blurb about the lock shows up every 10 seconds when run with `--debug`. I've wiped `~/.gradle` and rebooted but none of that helped.

FWIW the simulator window will open up when I go via the broken VSCode method.

[image]

ThadHouse, 2021-11-19:
Ah, youre on a mac (sorry I forgot). Theres another bug on there causing that task to not work on mac. So the workaround there is to take the command that is getting ran, and move the `-XstartOnFirstThread` argument to before `-jar`

swalberg, 2021-11-19:
This just isn't my night...

```
sean~/FRCProjects/romi5$ /Users/sean/wpilib/2022/jdk/bin/java -Djava.library.path=/Users/sean/FRCProjects/romi5/build/jni/release -Dfile.encoding=UTF-8 -Duser.country=US -Duser.language=en -Duser.variant -XstartOnFirstThread -jar /Users/sean/FRCProjects/romi5/build/libs/romi5.jar
java.io.IOException: wpiHaljni could not be loaded from path or an embedded resource.
	attempted to load for platform /osx/x86-64/
Last Load Error:
/Users/sean/FRCProjects/romi5/build/jni/release/libwpiHaljni.dylib: dlopen(/Users/sean/FRCProjects/romi5/build/jni/release/libwpiHaljni.dylib, 1): Library not loaded: libwpiHal.dylib
  Referenced from: /Users/sean/FRCProjects/romi5/build/jni/release/libwpiHaljni.dylib
  Reason: image not found

	at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
	at edu.wpi.first.hal.JNIWrapper.<clinit>(JNIWrapper.java:34)
	at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:424)
	at frc.robot.Main.main(Main.java:23)
```
I see the file is in the build directory...

```
sean~/FRCProjects/romi5$ export DYLD_LIBRARY_PATH=/Users/sean/FRCProjects/romi5/build/jni/release/
sean~/FRCProjects/romi5$ /Users/sean/wpilib/2022/jdk/bin/java -Djava.library.path=/Users/sean/FRCProjects/romi5/build/jni/release -Dfile.encoding=UTF-8 -Duser.country=US -Duser.language=en -Duser.variant -XstartOnFirstThread -jar /Users/sean/FRCProjects/romi5/build/libs/romi5.jar
HAL Extensions: No extensions found
********** Robot program starting **********
********** Robot program startup complete **********
Default simulationPeriodic() method... Override me!
Warning at edu.wpi.first.wpilibj.IterativeRobotBase.printLoopOverrunMessage(IterativeRobotBase.java:359): Loop time of 0.02s overrun
```
so that runs but the simulator window doesn't appear. Was there a second step I needed to do?

ThadHouse, 2021-11-19:
Got to export `HALSIM_EXTENSIONS` as well. It needs to point to libhalsim_gui.dylic, libhalsim_ws_client.dll, and libhalsim_ws_client.dll

Additonally, you'll need to set `HALSIMWS_HOST` to `10.0.0.2`

swalberg, 2021-11-19:
OK, I think I'm getting somewhere.

```
export HALSIM_EXTENSIONS=/Users/sean/FRCProjects/romi5/build/jni/release/libhalsim_gui.dylib:/Users/sean/FRCProjects/romi5/build/jni/release/libhalsim_ws_client.dylib:/Users/sean/FRCProjects/romi5/build/jni/release/libhalsim_ws_server.dylib
export HALSIMWS_HOST=10.0.0.1
export DYLD_LIBRARY_PATH=/Users/sean/FRCProjects/romi5/build/jni/release/
/Users/sean/wpilib/2022/jdk/bin/java -Djava.library.path=/Users/sean/FRCProjects/romi5/build/jni/release -Dfile.encoding=UTF-8 -Duser.country=US -Duser.language=en -Duser.variant -XstartOnFirstThread -jar /Users/sean/FRCProjects/romi5/build/libs/romi5.jar
```
The simulator starts up. I get good looking stuff on stdout including `Will attempt to connect to ws://10.0.0.1:3300/wpilibws`. And it would probably succeed if I hand't left my Romi at the build site. But I'll stop here now since I think this'll work. I'll grab my robot later and test it out.

Let me know if you want this ticket closed or left open until the release...

Thanks!

ThadHouse, 2021-11-19:
Yeah that should be everything necessary. We can leave it open for now, and we'll close it when everything pushes. Most of the fixes are still open in PR's

PeterJohnson, 2021-11-29:
Should be fixed in Beta 3, reopen if necessary after testing.

closed 2021-11-29 by PeterJohnson

swalberg, 2021-11-29:
I can't test with an actual Romi right now, but after installing Beta3 I was able to build the code and see that the simulator is trying to talk to the right host when going through VSCode, so I'm pretty sure this'll work. Thanks! 🎉 🎉 🎉