wpilibsuite/allwpilib #3737: DifferentialDriveVoltageConstraint seems to fail testing when using different voltage limits

jsimpso81, 2021-11-27:
When testing the JAVA version of the differential drive voltage constraint, the test fails if different voltage are used.  I don't know if this is a true failure (although it seems like one) or just an anomaly because I changed the max voltage value without changing the motor feedforward values.   

= = = = = = = = = = = = = = = = = = = = = = = = = = = =
voltage constraint  = 12.0
Test - **PASSED**
leftMaxExceeded = false
leftMinExceeded = false   
rightMaxExceeded = false  
rightMinExceeded = false  
leftMax = 9.956575185217648   
leftMin = -11.995226276701908 
rightMax = 9.95657560277206   
rightMin = -11.995226276701908
= = = = = = = = = = = = = = = = = = = = = = = = = = = =
voltage constraint  = 11.0
Test - **PASSED**
leftMaxExceeded = false   
leftMinExceeded = false   
rightMaxExceeded = false  
rightMinExceeded = false  
leftMax = 9.927802746621627   
leftMin = -10.998140822044537 
rightMax = 9.92781165132986   
rightMin = -10.998140822044537
= = = = = = = = = = = = = = = = = = = = = = = = = = = =
voltage constraint  = 10.0
Test - **PASSED**
leftMaxExceeded = false
leftMinExceeded = false
rightMaxExceeded = false
rightMinExceeded = false
leftMax = 9.236943640992521
leftMin = -9.99845187491805
rightMax = 9.355316198492204
rightMin = -9.998451874918054
= = = = = = = = = = = = = = = = = = = = = = = = = = = =
voltage constraint  = 9.0
Test - **FAILED**
leftMaxExceeded = true
leftMinExceeded = false
rightMaxExceeded = true
rightMinExceeded = false
leftMax = 9.179414074794652
leftMin = -8.99519873539483
rightMax = 9.358212156864694
rightMin = -8.995198735394835
= = = = = = = = = = = = = = = = = = = = = = = = = = = =
voltage constraint  = 8.0
Test - **FAILED**
leftMaxExceeded = true
leftMinExceeded = false
rightMaxExceeded = true
rightMinExceeded = false
leftMax = 8.574977407968161
leftMin = -7.999034759380962
rightMax = 9.354715680090512
rightMin = -7.999034759380966
= = = = = = = = = = = = = = = = = = = = = = = = = = = =
voltage constraint  = 7.0
Test - **FAILED**
leftMaxExceeded = true
leftMinExceeded = false
rightMaxExceeded = true
rightMinExceeded = false
leftMax = 8.1712315228454
leftMin = -6.9995891270447075
rightMax = 9.37143570562599
rightMin = -6.999589127044711

**To Reproduce**
[... 175 lines cut ...]
                new ArrayList<Translation2d>(),
                new Pose2d(0, 1, Rotation2d.fromDegrees(180)),
                config);

        TrajectoryGenerator.generateTrajectory(
                new Pose2d(0, 1, Rotation2d.fromDegrees(180)),
                new ArrayList<Translation2d>(),
                new Pose2d(1, 0, Rotation2d.fromDegrees(90)),
                config.setReversed(true));
    }
    }

**Expected behavior**
I thought that the tests would pass.

**Screenshots**
The test results are shown above.

**Desktop (please complete the following information):**
 - WPILib Version: [.2021.3.1]
 - OS: [e.g. Windows 10]
 - Java version [uncertain]  whatever came with the 2021 visual studio release
 - C++ version [uncertain]

**Additional context**

calcmogul, 2021-11-27:
As per the source-level comment, the test's voltage check is technically incorrect since it's using chassis acceleration instead of wheel acceleration.

jsimpso81, 2021-11-28:
So I modified the test program to calculate the acceleration using the previous and next samples. I created a variant that widened the time by using a previous and next sample that were 2 dt away.  Just for fun I added a kinematics constraint.  -- Unless I messed the accel calc up, which is entirely likely -- there still seemed to be an issue with this constraint.  I didn't start to look for the issue, but I'd probably start with the trajectory parameterizer.  I attached the test program and a spreadsheet plotting the results.

(After thinking about this for a while, it would seem to make sense that the individual wheel accel is more aggressive that the chassis accel.)

[DiffVoltConstraint_TestResults.xlsx](https://github.com/wpilibsuite/allwpilib/files/7614110/DiffVoltConstraint_TestResults.xlsx)

[TestDiffVoltCont.txt](https://github.com/wpilibsuite/allwpilib/files/7614112/TestDiffVoltCont.txt)

sciencewhiz, 2021-11-29:
I'm not sure how it would be possible for chassis acceleration to exceed without at least one wheel acceleration to also exceed

calcmogul, 2021-11-29:
It isn't.

jsimpso81, 2021-11-29:
I agree.  That is why when the original test failed with a different voltage.  I was fairly certain it would fail if I calculated wheel accel.   ( I added the extra sample mostly to see the data.)