wpilibsuite/allwpilib #3747: Control word used by various library components should be updated atomically

calcmogul, 2021-11-29:
In 2021, IterativeRobotBase and CommandScheduler both used DriverStation functions to check the control word. Even though the DS thread caches it, the data being asynchronously updated in the DS thread means a race condition occur between when IterativeRobotBae checks the DS functions and when CommandScheduler does.

In 2022, IterativeRobotBase was rewritten to check the control word directlly, so the cached DS control word can get out of sync with the control word in IterativeRobotBase.

One solution would be to move the control word updates out of the DS thread and into IterativeRobotBase. The updates would then happen synchronously. The problem with that approach is the user can block the main thread, which would prevent control word updates. The EducationalRobot example would also need to include the control word update since it inherits from RobotBase. The control word move wouldn't affect joystick updates or MotorSafety in the DS thread though.

chauser, 2021-11-30:
I see the issue has been re-opened. One concern is that the cache timeout in DriverStation.updateControlWord is hard-coded 50ms. I've learned to be suspicious of hard-coded timeouts -- they suggest a design decision that was not carefully thought through; in this case, is a 50ms timeout appropriate for a 5, 10, or 20ms robot period? I don't know, but it doesn't feel right to me.
Maybe it doesn't matter any more, now that the main loop is early-on forcing a cache update. But it still caught my attention.

ThadHouse, 2021-11-30:
That time is just if something hasn't passed true to the function to force an update. The function called from the robot base does pass true, so it will force an update.

ThadHouse, 2021-11-30:
The timeout actually wasn't the root cause of the bug BTW. The root cause was the forced cache update in the DS loop happened after the new data callback in the DS thread was triggered. So if the robot loop started after the DS packet was received, but before the DS thread woke up to process the new data, the robot loop would act enabled, but the DS cache wouldn't have been updated, so anything hitting it would see disabled. By moving the robot loop to also using the cache, this  bug is fixed in the Disabled->Enabled state.

Note that using the cache just moves the bug from Disabled->Enabled to Enabled->Disabled. But that is much less likely to cause a problem, other then maybe an extra command loop. and is probably unobservable. It's fixable, but requires a LARGE change its too late to do for 2022.

chauser, 2021-11-30:
Thanks for the explanation. It all makes sense now.

calcmogul, 2021-12-01:
For 2023, we're planning on addressing this issue by moving the control word and joystick data updates in the DS thread to a DriverStation::UpdateDriverStation() function that's called synchronously by IterativeRobotBase::LoopFunc().

The only thing left in the DS thread would be a call to MotorSafety::CheckMotors(). We'd move this thread to the MotorSafety class.

Starlight220, 2022-10-30:
What's the status of this, now that #3787 was merged?

ThadHouse, 2023-08-03:
This has been fixed by the DS changes done last season.

closed 2023-08-03 by ThadHouse