wpilibsuite/allwpilib #3765: MecanumControllerCommand & SwerveControllerCommand examples crash when enabling autonomous mode in simulation

Modelmat, 2021-12-06:
**Describe the bug**
MecanumControllerCommand & SwerveControllerCommand examples crash when enabling autonomous mode in simulation.

This behaviour occurs in at least C++, but may or may not occur in Java, as well.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the above commands in simulation.
2. Switch to the "Autonomous" mode.

**Expected behavior**
No crash, and probably working however the example is intended.

**Backtraces**
*Mecanum*
```cpp
Warning at Main: Warning: Watchdog not fed within 0.020000s
	at frc::Watchdog::Impl::Main() + 0x231 [0x7ffff7ecef31]
	at  + 0xd6de4 [0x7ffff78a4de4]
	at  + 0x9609 [0x7ffff764a609]
	at clone + 0x43 [0x7ffff7571293]

Error at PrintLoopOverrunMessage: Error: Loop time of 0.020000s overrun
	at frc::IterativeRobotBase::PrintLoopOverrunMessage() + 0x56 [0x7ffff7eb9426]
	at frc::Watchdog::Impl::Main() + 0x127 [0x7ffff7ecee27]
	at  + 0xd6de4 [0x7ffff78a4de4]
	at  + 0x9609 [0x7ffff764a609]
	at clone + 0x43 [0x7ffff7571293]

CommandScheduler loop time overrun.
Error at Check: A timeout has been exceeded: MecanumDrive... Output not updated often enough
	at frc::MotorSafety::Check() + 0x100 [0x7ffff7eb9de0]
	at frc::MotorSafety::CheckMotors() + 0x8d [0x7ffff7eb9ead]
	at  + 0xa61f1 [0x7ffff7ea51f1]
	at  + 0xd6de4 [0x7ffff78a4de4]
	at  + 0x9609 [0x7ffff764a609]
	at clone + 0x43 [0x7ffff7571293]

Thread 19 "MecanumControll" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffa6e98700 (LWP 4405)]
0x00007ffff7f906de in frc2::MecanumControllerCommand::<lambda()>::operator() (__closure=0x7fff9801fc20) at ../wpimath/src/main/native/include/frc/geometry/Pose2d.h:112
112	  const Rotation2d& Rotation() const { return m_rotation; }
(gdb) bt
#0  0x00007ffff7f906de in frc2::MecanumControllerCommand::<lambda()>::operator() (__closure=0x7fff9801fc20) at ../wpimath/src/main/native/include/frc/geometry/Pose2d.h:112
#1  std::_Function_handler<frc::Rotation2d(), frc2::MecanumControllerCommand::MecanumControllerCommand(frc::Trajectory, std::function<frc::Pose2d()>, frc::SimpleMotorFeedforward<units::unit<std::ratio<1>, units::base_unit<std::ratio<1> > > >, frc::MecanumDriveKinematics, frc2::PIDController, frc2::PIDController, frc::ProfiledPIDController<units::unit<std::ratio<1>, units::base_unit<std::ratio<0, 1>, std::ratio<0, 1>, std::ratio<0, 1>, std::ratio<1> > > >, units::velocity::meters_per_second_t, std::function<frc::MecanumDriveWheelSpeeds()>, frc2::PIDController, frc2::PIDController, frc2::PIDController, frc2::PIDController, std::function<void(units::unit_t<units::unit<std::ratio<1>, units::base_unit<std::ratio<2>, std::ratio<1>, std::ratio<-3>, std::ratio<0, 1>, std::ratio<-1> > > >, units::unit_t<units::unit<std::ratio<1>, units::base_unit<std::ratio<2>, std::ratio<1>, std::ratio<-3>, std::ratio<0, 1>, std::ratio<-1> > > >, units::unit_t<units::unit<std::ratio<1>, units::base_unit<std::ratio<2>, std::ratio<1>, std::ratio<-3>, std::ratio<0, 1>, std::ratio<-1> > > >, units::unit_t<units::unit<std::ratio<1>, units::base_unit<std::ratio<2>, std::ratio<1>, std::ratio<-3>, std::ratio<0, 1>, std::ratio<-1> > > >)>, std::initializer_list<frc2::Subsystem*>)::<lambda()> >::_M_invoke(const std::_Any_data &) (__functor=...) at /usr/include/c++/9/bits/std_function.h:285
#2  0x00007ffff7f90874 in std::function<frc::Rotation2d ()>::operator()() const (this=0x7fff9801fc20) at /usr/include/c++/9/bits/std_function.h:683
#3  frc2::MecanumControllerCommand::Execute (this=0x7fff9801f7b0) at ../wpilibNewCommands/src/main/native/cpp/frc2/command/MecanumControllerCommand.cpp:285
#4  0x00007ffff7f9f931 in frc2::SequentialCommandGroup::Execute (this=0x7fff9801cb60) at /usr/include/c++/9/bits/unique_ptr.h:360
#5  frc2::SequentialCommandGroup::Execute (this=0x7fff9801cb60) at ../wpilibNewCommands/src/main/native/cpp/frc2/command/SequentialCommandGroup.cpp:24
#6  0x00007ffff7f8da37 in frc2::CommandScheduler::Run (this=0x7ffff7fc69a0 <frc2::CommandScheduler::GetInstance()::scheduler>) at ../wpilibNewCommands/src/main/native/cpp/frc2/command/CommandScheduler.cpp:220
#7  0x00007ffff7eb902d in frc::IterativeRobotBase::LoopFunc (this=0x55555557d200 <frc::impl::RunRobot<Robot>(std::mutex&, Robot**)::theRobot>) at ../wpilibc/src/main/native/cpp/IterativeRobotBase.cpp:172
#8  0x00007ffff7eca3c4 in std::function<void ()>::operator()() const (this=0x7fffa6e97b50) at /usr/include/c++/9/bits/std_function.h:683
#9  frc::TimedRobot::StartCompetition (this=0x55555557d200 <frc::impl::RunRobot<Robot>(std::mutex&, Robot**)::theRobot>) at ../wpilibc/src/main/native/cpp/TimedRobot.cpp:50
#10 0x000055555556df41 in frc::impl::RunRobot<Robot> (m=..., robot=0x55555557d9f8 <frc::StartRobot<Robot>()::robot>) at /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:779
#11 0x000055555556e0fd in frc::StartRobot<Robot>()::{lambda()#1}::operator()() const (this=<optimised out>) at ../wpilibc/src/main/native/include/frc/RobotBase.h:65
#12 std::__invoke_impl<void, frc::StartRobot<Robot>()::{lambda()#1}>(std::__invoke_other, frc::StartRobot<Robot>()::{lambda()#1}&&) (__f=...) at /usr/include/c++/9/bits/invoke.h:60
#13 std::__invoke<frc::StartRobot<Robot>()::{lambda()#1}>(std::__invoke_result&&, (frc::StartRobot<Robot>()::{lambda()#1}&&)...) (__fn=...) at /usr/include/c++/9/bits/invoke.h:95
#14 std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=<optimised out>) at /usr/include/c++/9/thread:244
#15 std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> >::operator()() (this=<optimised out>) at /usr/include/c++/9/thread:251
#16 std::thread::_State_impl<std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> > >::_M_run() (this=<optimised out>) at /usr/include/c++/9/thread:195
#17 0x00007ffff78a4de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#18 0x00007ffff764a609 in start_thread (arg=<optimised out>) at pthread_create.c:477
#19 0x00007ffff7571293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
*Swerve*
```cpp
Default SimulationPeriodic() method... Override me!
[Thread 0x7fffa6697700 (LWP 4628) exited]
Warning at Main: Warning: Watchdog not fed within 0.020000s
	at frc::Watchdog::Impl::Main() + 0x231 [0x7ffff7ecef31]
[... 26 lines cut ...]
#5  frc2::SequentialCommandGroup::Execute (this=0x7fff9802a200) at ../wpilibNewCommands/src/main/native/cpp/frc2/command/SequentialCommandGroup.cpp:24
#6  0x00007ffff7f8da37 in frc2::CommandScheduler::Run (this=0x7ffff7fc69a0 <frc2::CommandScheduler::GetInstance()::scheduler>) at ../wpilibNewCommands/src/main/native/cpp/frc2/command/CommandScheduler.cpp:220
#7  0x00007ffff7eb902d in frc::IterativeRobotBase::LoopFunc (this=0x555555582080 <frc::impl::RunRobot<Robot>(std::mutex&, Robot**)::theRobot>) at ../wpilibc/src/main/native/cpp/IterativeRobotBase.cpp:172
#8  0x00007ffff7eca3c4 in std::function<void ()>::operator()() const (this=0x7fffa6e97b50) at /usr/include/c++/9/bits/std_function.h:683
#9  frc::TimedRobot::StartCompetition (this=0x555555582080 <frc::impl::RunRobot<Robot>(std::mutex&, Robot**)::theRobot>) at ../wpilibc/src/main/native/cpp/TimedRobot.cpp:50
#10 0x0000555555566ea1 in frc::impl::RunRobot<Robot> (m=..., robot=0x555555583288 <frc::StartRobot<Robot>()::robot>) at /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:779
#11 0x000055555556705d in frc::StartRobot<Robot>()::{lambda()#1}::operator()() const (this=<optimised out>) at ../wpilibc/src/main/native/include/frc/RobotBase.h:65
#12 std::__invoke_impl<void, frc::StartRobot<Robot>()::{lambda()#1}>(std::__invoke_other, frc::StartRobot<Robot>()::{lambda()#1}&&) (__f=...) at /usr/include/c++/9/bits/invoke.h:60
#13 std::__invoke<frc::StartRobot<Robot>()::{lambda()#1}>(std::__invoke_result&&, (frc::StartRobot<Robot>()::{lambda()#1}&&)...) (__fn=...) at /usr/include/c++/9/bits/invoke.h:95
#14 std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=<optimised out>) at /usr/include/c++/9/thread:244
#15 std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> >::operator()() (this=<optimised out>) at /usr/include/c++/9/thread:251
#16 std::thread::_State_impl<std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> > >::_M_run() (this=<optimised out>) at /usr/include/c++/9/thread:195
#17 0x00007ffff78a4de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#18 0x00007ffff764a609 in start_thread (arg=<optimised out>) at pthread_create.c:477
#19 0x00007ffff7571293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

**Desktop (please complete the following information):**
 - WPILib Version: [e.g. 2021.3.1] master
 - OS: [e.g. Windows 11] ElementaryOS 6, Kernel 5.11 (Ubuntu derivative)
 - Java version [e.g. 1.10.2] n/a
 - C++ version [e.g. 17] 17? whatever the default is

**Additional context**
N/A

PeterJohnson, 2021-12-07:
The issue is that a lambda capture is used in a command constructor: https://github.com/wpilibsuite/allwpilib/blob/main/wpilibNewCommands/src/main/native/cpp/frc2/command/MecanumControllerCommand.cpp#L84-L86 (also 251-253).

This is unsafe in C++ as the command can be moved after creation (which changes the captured "this" pointer).  This lambda setting must be delayed until Initialize().

Modelmat, 2021-12-07:
Welp, same problem different example, lol.

https://github.com/wpilibsuite/allwpilib/issues/3109#issuecomment-763396949

[image]

sciencewhiz, 2021-12-17:
After moving the lambda to Initialize() in the SwerveControllerCommand example, it crashes after ~5 seconds of running in autonomous.

```
ucrtbased.dll!00007ffb0890afec() (Unknown Source:0)
ucrtbased.dll!00007ffb0890af37() (Unknown Source:0)
wpimathd.dll!std::vector<frc::Trajectory::State,std::allocator<frc::Trajectory::State>>::front() Line 1521 (c:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\vector:1521)
wpimathd.dll!frc::Trajectory::Sample(units::unit_t<units::unit<std::ratio<1,1>,units::base_unit<std::ratio<0,1>,std::ratio<0,1>,std::ratio<1,1>,std::ratio<0,1>,std::ratio<0,1>,std::ratio<0,1>,std::ratio<0,1>,std::ratio<0,1>,std::ratio<0,1>>,std::ratio<0,1>,std::ratio<0,1>>,double,units::linear_scale> t) Line 70 (c:\Users\Joe\Documents\Robotics\allwpilib\wpimath\src\main\native\cpp\trajectory\Trajectory.cpp:70)
frcUserProgram.exe!frc2::SwerveControllerCommand<4>::Execute() Line 97 (c:\Users\Joe\.gradle\caches\transforms-3\2afc89ad23ad6f6d508aac4983ac9a57\transformed\wpilibNewCommands-cpp-2022.424242.1.1-beta-3-20211216154507-29-gc5ae0ef-dirty-headers\frc2\command\SwerveControllerCommand.inc:97)
wpilibNewCommandsd.dll!frc2::SequentialCommandGroup::Execute() Line 32 (c:\Users\Joe\Documents\Robotics\allwpilib\wpilibNewCommands\src\main\native\cpp\frc2\command\SequentialCommandGroup.cpp:32)
wpilibNewCommandsd.dll!frc2::CommandScheduler::Run() Line 221 (c:\Users\Joe\Documents\Robotics\allwpilib\wpilibNewCommands\src\main\native\cpp\frc2\command\CommandScheduler.cpp:221)
frcUserProgram.exe!Robot::RobotPeriodic() Line 22 (c:\Users\Joe\Documents\Robotics\2022Beta3SwerveControllerCommandCPP\src\main\cpp\Robot.cpp:22)
wpilibcd.dll!frc::IterativeRobotBase::LoopFunc() Line 173 (c:\Users\Joe\Documents\Robotics\allwpilib\wpilibc\src\main\native\cpp\IterativeRobotBase.cpp:173)
wpilibcd.dll!frc::TimedRobot::{ctor}::__l2::<lambda>() Line 77 (c:\Users\Joe\Documents\Robotics\allwpilib\wpilibc\src\main\native\cpp\TimedRobot.cpp:77)
wpilibcd.dll!std::invoke<void <lambda>(void) &>(frc::TimedRobot::{ctor}::__l2::void <lambda>(void) & _Obj) Line 1479 (c:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\type_traits:1479)
wpilibcd.dll!std::_Invoker_ret<void,1>::_Call<void <lambda>(void) &>(frc::TimedRobot::{ctor}::__l2::void <lambda>(void) & _Func) Line 652 (c:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\functional:652)
wpilibcd.dll!std::_Func_impl_no_alloc<void <lambda>(void),void>::_Do_call() Line 823 (c:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\functional:823)
wpilibcd.dll!std::_Func_class<void>::operator()() Line 870 (c:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\functional:870)
wpilibcd.dll!frc::TimedRobot::StartCompetition() Line 52 (c:\Users\Joe\Documents\Robotics\allwpilib\wpilibc\src\main\native\cpp\TimedRobot.cpp:52)
frcUserProgram.exe!frc::impl::RunRobot<Robot>(std::mutex & m, Robot * * robot) Line 34 (c:\Users\Joe\.gradle\caches\transforms-3\3bcfdc68792232a25d3100be3966fd68\transformed\wpilibc-cpp-2022.424242.1.1-beta-3-20211216154507-29-gc5ae0ef-dirty-headers\frc\RobotBase.h:34)
frcUserProgram.exe!frc::StartRobot::__l8::<lambda>() Line 67 (c:\Users\Joe\.gradle\caches\transforms-3\3bcfdc68792232a25d3100be3966fd68\transformed\wpilibc-cpp-2022.424242.1.1-beta-3-20211216154507-29-gc5ae0ef-dirty-headers\frc\RobotBase.h:67)
frcUserProgram.exe!std::invoke<void <lambda>(void)>(frc::StartRobot::__l8::void <lambda>(void) && _Obj) Line 1479 (c:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\type_traits:1479)
frcUserProgram.exe!std::thread::_Invoke<std::tuple<void <lambda>(void)>,0>(void * _RawVals) Line 56 (c:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\thread:56)
ucrtbased.dll!00007ffb08914c7c() (Unknown Source:0)
```

sciencewhiz, 2021-12-18:
The crash above is caused by the example calling swerveControllerCommand with the same trajectory twice. Presumably the crash is called because of a discontinuity since there are different starting and ending points

PR #3795 (2021-12-18): Fix Mecanum & Swerve ControllerCommand lamda capture
Fixes #3765
Also fixes SwerveControllerCommand example calling command twice
commit d41d051f (2021-12-18): [wpilibc] Fix Mecanum & Swerve ControllerCommand lambda capture (#3795)

Fixes #3765
Also fixes SwerveControllerCommand example calling command twice.
closed 2021-12-18 by PeterJohnson