wpilibsuite/allwpilib #3771: Remove getAverageEncoderDistance from all examples

jasondaming, 2021-12-06:
Getting the encoder average between two sides of a differential drivetrain does not help and it can hurt.  What should be done is to look for the encoder that has the highest velocity.

joshuathooyavan, 2021-12-18:
Why would you do that?

calcmogul, 2021-12-18:
The most common failure mode for an encoder is it no longer counting, which means the reported velocity will be zero. If the magnitude of the other encoder is nonzero, it's more likely to be working.

jasondaming, 2021-12-18:
@joshuathooyavan averaging the encoders is a trap many fall into.  It is actually one of the worst things you can do, because if one of them fails all of your encoders will be reading half their expected value.  There is also a common misconception that it improves accuracy.  This is simply not true as you don't know which encoder is correct and averaging them just ensures you have an incorrect answer.

Oblarg, 2021-12-27:
The standard error of the average is smaller than the standard error of the individual measurement by a factor of sqrt(2).  That *is* an improvement.  Fault checking should be done at a different level of abstraction.