wpilibsuite/allwpilib #3781: CameraServer 2022 memory crash

noamzaks, 2021-12-11:
**We are currently trying to get a minimal working example of this issue with our roboRIO, then report back.**

**Describe the bug**
Using cameraserver seems to crash the robot code because of string_view issues.

The robot code is stuck in a restart loop.

**To Reproduce**
Steps to reproduce the behavior:
1. ...
2. ...

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - WPILib Version: 2022.1.1-beta-2
 - OS: Windows 11
 - Java version 
 - C++ version 17

PeterJohnson, 2021-12-11:
We made a fix in this area right before the beta 3 release. Please try beta 3 and see if it fixes your issue.

noamzaks, 2021-12-11:
Will do

noamzaks, 2021-12-14:
Issue seems to be resolved. 
Thanks to @liorsagy for verifying.

closed 2021-12-14 by noamzaks