wpilibsuite/allwpilib #3810: Consider adding RomiGyro class to Romi templates

agasser, 2021-12-20:
The RomiReference example project includes a `RomiGyro` class that implements gyro support for the Romi. This class knows internals about the way the Romi communicates with the simulator to provide gyro data that a user could no know.

Since this class and the knowledge it has is not documented anywhere in the WPILib documentation, this class should be included in the "Romi - Timed Robot" and "Romi - Command Robot" templates.

An alternative approach would be to add this class to WPILib or to a Romi vendor library. Another alternative approach would be to include the knowledge this class requires in the WPILib documentation so a user could recreate the class from scratch.

Starlight220, 2021-12-20:
It's included in a Romi vendordep, here's the JSON: https://github.com/wpilibsuite/romi-vendordep/blob/main/RomiVendordep.json

agasser, 2021-12-20:
Is this documented? I would be willing to add it to the Romi Getting Started documentation section, but I want to be sure I'm adding accurate information.

agasser, 2021-12-20:
Given that these are included in a vendordep, the example should be updated to reference the vendordep instead of having the `RomiGyro` implementation in the project. Additionally, the templates should be updated to include the vendordep.
If this sounds OK, I would be willing to send a PR.

agasser, 2021-12-21:
> It's included in a Romi vendordep, here's the JSON: https://github.com/wpilibsuite/romi-vendordep/blob/main/RomiVendordep.json

Using that JSON as an online vendor library does not work, the artifacts are not published to Maven.

ThadHouse, 2021-12-21:
Ah, the artifacts are not published to release maven, just development. Thats something that needs to be fixed.

Starlight220, 2022-09-20:
Was this fixed?

sciencewhiz, 2022-09-21:
The Romi vendordep publishing was fixed.
The vendordep was documented in frc-docs. https://github.com/wpilibsuite/frc-docs/pull/1621
But the main issue has not been done. It requires changes in both allwpilib to the Romi examples to use the vendordep (https://github.com/wpilibsuite/allwpilib/pull/3870) It also requires changes in the VSCode extension (https://github.com/wpilibsuite/vscode-wpilib/issues/528 and https://github.com/wpilibsuite/vscode-wpilib/issues/508)

closed 2023-09-19 by PeterJohnson