wpilibsuite/allwpilib #3813: EncoderSim has differing methods between Java/C++

modelmat, 2021-12-21:
**Describe the bug**
In Java, setDistancePerPulse/getDistancePerPulse do not exist. In C++, they do.
There may be more methods which have this property.

**To Reproduce**
https://github.com/wpilibsuite/allwpilib/blob/9f8f330e965fb0c8c05b2c53c1f65a84a2dfcaaa/wpilibc/src/main/native/cpp/simulation/EncoderSim.cpp#L178-L180
https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/src/main/java/edu/wpi/first/wpilibj/simulation/EncoderSim.java

**Expected behavior**
Either both Java & C++ don't have the method (probably makes more sense, since Encoder stores the distancePerPulse in the class itself, and it's not a physical measurement), or they both do.

**Desktop (please complete the following information):**
 - WPILib Version: latest

Starlight220, 2021-12-24:
Do the methods exist at the JNI level?

modelmat, 2021-12-28:
I believe that's a no

https://github.com/wpilibsuite/allwpilib/tree/main/hal/src/main/native/cpp/jni/simulation/EncoderDataJNI.cpp

PR #4928 (2023-01-11): Add missing distance per pulse functions to java EncoderSim
Also fix C++ and Java `EncoderSim.setDistancePerPulse()` not propagating value to `SimEncoderData`

Closes #3813
closed 2023-01-11 by PeterJohnson