wpilibsuite/allwpilib #3825: HolonomicDriveController.java -- Should Calculate reset PID when not enabled

jsimpso81, 2021-12-23:
**Describe the bug**
Should the Calculate method of the holonomicDriveController reset the PIDControllers when not enabled.

    if (!m_enabled) {
      return ChassisSpeeds.fromFieldRelativeSpeeds(xFF, yFF, thetaFF, currentPose.getRotation());
      // -------- add 
      m_xController.reset;
      m_yController.reset;
      //-------- end of suggestions
    }

**To Reproduce**
Steps to reproduce the behavior:
1. Just found when reviewing code.  I didn't do any testing.

**Expected behavior**
- If the holonomic controller is disabled, when reenabled the PID controllers may have different integral characteristics causing an error in the control.  It seems like the PID control should be reset -- It would be even better if the PID controller had tracking capability so it wouldn't bump when re-enabled  (That is a whole different topic.).

**Screenshots**
- None
-
**Desktop (please complete the following information):**
 - Checked with latest source code as of 12/22

**Additional context**
- None

calcmogul, 2021-12-23:
If it does get reset, it should only be done when transitioning to disabled, not in every call to calculate().

jsimpso81, 2021-12-23:
Agreed.  Sorry, I wasn't really thinking straight.

calcmogul, 2023-08-03:
If the PID controller has an integral state appropriately set to fix steady-state error, resetting it on disable would clear that and make the response worse when enabling again. If users want reset behavior, they should call it manually via the PIDController getters.

closed 2023-08-03 by calcmogul