wpilibsuite/allwpilib #3827: RamseteCommand example (and maybe others) motor inversion

agasser, 2021-12-24:
The [`DriveSubsystem` in the RamseteCommand example](https://github.com/wpilibsuite/allwpilib/blob/main/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/ramsetecommand/subsystems/DriveSubsystem.java) has not been updated completely for the right motor controller inversion breaking change to `DifferentialDrive`. The `rightVolts` parameter is incorrectly inverted in the `tankDriveVolts()` method. This may impact other examples, I did not check others.

Steps to reproduce the behavior:
1. Run the RamseteCommand example.
2. Observe that driving the robot in teleop functions properly, but attempting to drive a trajectory results in unpredictable robot movement that usually ends with the robot spinning in a circle.

Proposed fix:
Do _not_ invert the `rightVolts` parameter in the `tankDriveVolts()` method.

agasser, 2021-12-24:
This also impacts the trajectory tutorial documentation because it references (an out-of-date version) of the RamseteCommand example.
See https://github.com/wpilibsuite/frc-docs/issues/1577

Daltz333, 2021-12-24:
I was just looking to open an issue and saw that you already had one!

PR #3836 (2021-12-27): Ensure right side motors are inverted in examples
Fixes #3827
Adds MotorController inversion for right side, removes inversion in
setVoltage methods.
commit dceb5364 (2021-12-27): [examples] Ensure right side motors are inverted (#3836)

Fixes #3827
Adds MotorController inversion for right side, removes inversion in
setVoltage methods.

Also fixes various XboxController negations (was inconsistent throughout examples).
closed 2021-12-27 by PeterJohnson